FortySevenEffects/arduino_midi_library #149: class usbMidi::usbMidiTransport' has no member named 'thruActivated'

deladriere, 2020-04-26:
when I do 
```
#include <USB-MIDI.h>
USBMIDI_CREATE_DEFAULT_INSTANCE();
```

I get this error
```
MIDI.hpp:97:34: error: 'class usbMidi::usbMidiTransport' has no member named 'thruActivated'
     mThruActivated  = mTransport.thruActivated;
```
But if I edit MIDI.hpp and comment that line: it compiles ok

lathoub, 2020-04-26:
Thanks for signalling @deladriere. Somehow the thruActivated was not picked up in the v1.1.1 release of the USB-MIDI library. I fixed it in release v1.1.2 and uploaded to GitHub (should propagate to Arduino IDE within 24hrs).

deladriere, 2020-05-02:
Ok thanks, it's working fine

closed 2020-05-02 by deladriere

YANG-Mingcong, 2021-01-20:
Still get the same Issue

YANG-Mingcong, 2021-01-20:
Cause in Arduino, MIDI-USB 1.1.1 version has been set as depended library. I suggest change it to 1.1.2.
Thanks a lot.