FortySevenEffects/arduino_midi_library #152: setHandleTimeCodeQuarterFrame returns status byte

MIDIMinion, 2020-05-16:
I'm working on a project that displays MIDI Time Code using a Teensy 4.0. When I start running time code, the display chases MTC for about 1.5 seconds before going haywire. After some troubleshooting, I noticed that the callback `setHandleTimeCodeQuarterFrame()` will occasionally pass a status byte instead of a data byte, and I believe that's what's throwing off my program.

Has anyone run into this before?

franky47, 2020-05-16:
Do you have a dump of the MIDI stream so I can run some tests ?

MIDIMinion, 2020-05-16:
Here's the MIDI stream data bytes post `setHandleTimeCodeQuarterFrame().` Timecode starts at 04:56:40;12 @ 24fps and runs for 2 seconds
[setHandleTimeCodeQuarterFrame 2 sec MIDI Stream.txt]

(https://github.com/FortySevenEffects/arduino_midi_library/files/4638446/setHandleTimeCodeQuarterFrame.2.sec.MIDI.Stream.txt)

and in case it's helpful, here's the code from my .ino

[Teensy MTC Tests code.txt](https://github.com/FortySevenEffects/arduino_midi_library/files/4638448/Teensy.MTC.Tests.code.txt)

franky47, 2020-05-16:
Looks like you're using the serial port for both Serial debugging (printing stuff to the console) and the MIDI stream, does the problem occur if the MIDI stream has its own dedicated serial port ?

MIDIMinion, 2020-05-16:
Ok, I think I managed to change the MIDI serial over to Serial2. Here's the resulting stream and .ino:

[setHandleTimeCodeQuarterFrame 2 sec MIDI Stream v2.txt](https://github.com/FortySevenEffects/arduino_midi_library/files/4638484/setHandleTimeCodeQuarterFrame.2.sec.MIDI.Stream.v2.txt)

[Teensy MTC Tests code v2.txt](https://github.com/FortySevenEffects/arduino_midi_library/files/4638485/Teensy.MTC.Tests.code.v2.txt)

looks like I'm still getting status bytes

mink99, 2020-05-18:
From the log it looks as if the some messages are getting lost, which may look as if „wrong data“ is read. In your callback routine you are updating the display . Updating a display takes ages, so while the update is running, callbacks may not be processed, aka race condition. 
You may try to move the display stuff into the main loop, by Duplicatin the data parameter from the callback and use the duplicate in your display code.  This way you may lose a screen refresh, but not a callback ....

MIDIMinion, 2020-05-20:
Thanks @mink99  and @franky47  ! It took some working out, but the process of writing to the display was the cause of the missing bytes and I was able to get that chunk of code working as expected. Thanks again for your help!

closed 2020-05-21 by franky47