FortySevenEffects/arduino_midi_library #156: [Question] Does MIDI Thru happen at the end of MIDI.read()?

MIDIMinion, 2020-05-20:
Based on what I've seen from testing my code/setup, when thru is enabled, the message is received then transmitted as the very last thing that happens when `MIDI.read()` catches an incoming MIDI message. Is this the case?

Is there any way to set the thru-put of the incoming MIDI message to be the very first thing to happen when `MIDI.read()` has a message come in? The code I'm working on has a need to display information, wait a period of time, and then display other information. and I'm noticing that after that period time has passed is when the message gets transmitted via Thru

franky47, 2020-05-21:
The way thru works is by parsing the MIDI stream, and sending valid messages once they are detected.

There are multiple reasons for that, but the main one is to allow merging the output stream (whatever your program sends) with the thru stream, without interlacing messages. It also allows filtering what is thru'd or not.

If you need a faster thru that ignores the output stream, I would suggest using a hardware thru:
[image]

franky47, 2020-05-21:
In your case, you could avoid active waiting and use a polling approach: instead of blocking the main loop, you periodically check for a flag that indicates your operation is done.

The program will be more complex (it may involve state machines and working with interrupts) but it will solve your performance problems.

MIDIMinion, 2020-05-22:
thanks for the insight! I've managed to get the program working with flags like you suggested. You were right in that it's a bit more complex, but in my case it's not too bad.

I appreciate your speedy responses, especially to little problems like mine! Great library as well, I'm really enjoying working with it! Thanks again!

closed 2020-05-22 by MIDIMinion