FortySevenEffects/arduino_midi_library #160: USB Midi hang when connected computer hibernates

SandwichRising, 2020-07-02:
So I have another unusual issue. I am again using an ATmega32U4. It is connected to a PC USB port and everything works as intended until I hibernate the computer. The MIDI device keeps receiving power from the computer and begins hanging on USB sends while the computer is hibernated, and when it's powered back on it acts normal again once the Windows login screen loads. The hangs cause around 1/4 second delay before the program loop can proceed, and the pro micro's LED which lights up on a USB send stays lit the entire time the computer is hibernated. 

If I comment out the USB MIDI .begin() as well as the USB sends, the computer does not keep sending power to the device when it goes into hibernation, powering the device off as it hibernates. 

If I power cycle the device while the PC is hibernated, it will power on and will work as normal. After the PC is powered back on and the ATmega32U4 communicates with the computer it will hang again if the computer is hibernated (it will actually communicate for a brief time during the PC's boot sequence and as soon as the boot proceeds on, it will begin hanging for the rest of the boot until Windows comes up).

Is there a way to tell the device to only send USB data if there is an established connection to prevent the hanging?

franky47, 2020-07-03:
This looks like an enumeration/de-enumeration issue, any ideas @lathoub ?

This issue probably belongs in the [USB-MIDI](https://github.com/lathoub/Arduino-USBMIDI) repository though, @SandwichRising can you transfer it there please ?

closed 2020-07-03 by franky47

SandwichRising, 2020-07-03:
Moved it over, thank you!