FortySevenEffects/arduino_midi_library #161: What happens to MIDI messages when I use MIDI.setInputChannel(MIDI_CHANNEL_OFF)?

MIDIMinion, 2020-07-14:
Do they get stored in the buffer? Is there a way to clear that buffer?

I'd like to be able to turn input on and off as I need, but I'm finding that when I go from input off to input on, and then read, it reads all of the messages that were received while input channel was turned off.

Is there another way of doing this?

Thanks!

franky47, 2020-07-14:
Yes, when the MIDI input is turned off, messages are not read and remain in the input buffer. You can flush it before  re-activating the MIDI input with this code:

```c++
while (Serial.available() > 0)
{
  Serial.read();
}
```

Replace `Serial` with the serial port used for MIDI (can be Serial1 on some boards, check out the [default value](https://github.com/FortySevenEffects/arduino_midi_library/issues/65))

MIDIMinion, 2020-07-14:
Ah! I had tried the serial.read() loop method with no success, but you've reminded me that I'm using Serial2! Works like a charm, thank you!

closed 2020-07-14 by MIDIMinion