FortySevenEffects/arduino_midi_library #163: MIDI Messages One Message Behind

acumartini, 2020-08-10:
Hey all. Not sure this is an issue with your library but want to confirm. I am running the Adafruit TinyUSB MIDI lib, which uses your lib under the hood and my messages appear to be one step behind.

For example the first message the system sees is a NOTE_ON message, but only after I **release** the note. So hitting three consecutive keys on a virtual midi keyboard shows up like this on the NoteOn and NoteOff handlers:

15:56:27.083 -> Note on: channel = 1, pitch = 60, velocity - 100
15:56:30.077 -> Note off: channel = 1, pitch = 60, velocity - 0
15:56:30.511 -> Note on: channel = 1, pitch = 62, velocity - 100
15:56:31.779 -> Note off: channel = 1, pitch = 62, velocity - 0
15:56:32.143 -> Note on: channel = 1, pitch = 65, velocity - 100

As you can see I released the last note and the last message is a NOTE_ON type. Any ideas or pointers? Should I be bugging the maintainers of the Adafruit lib?

Note I am on OSX using CoreMIDI to connect to the board via USB. Thanks!

franky47, 2020-10-05:
No idea what could happen here.. Did you have any luck with the Adafruit maintainers ?

Tpigman, 2020-11-01:
Any luck figuring this out?

touchgadget, 2021-12-17:
I ran into the problem using a Pi Pico board.

https://github.com/adafruit/Adafruit_TinyUSB_Arduino/issues/64 indicates the problem has been fixed.

https://github.com/earlephilhower/arduino-pico/issues/352 indicates if using the Earle Philhower RP2040 board package, it includes an out of date Adafruit_TinyUSB library. The solution is to install the latest version of the Adafruit_TinyUSB library using the library manager. The IDE will use the newer version.

franky47, 2021-12-18:
Thanks for the report @touchgadget, much appreciated.

Closing this for now, @acumartini feel free to reopen if the suggested fix does not help.

closed 2021-12-18 by franky47