FortySevenEffects/arduino_midi_library #164: TimeStamp to Messages

jmakurami09, 2020-08-11:
Does this library already allow for adding timestamps like JUCE or 

volumeSlider.onValueChange = [this]
        {
            auto message = juce::MidiMessage::controllerEvent (midiChannel, 7, (int) volumeSlider.getValue());
            **message.setTimeStamp (juce::Time::getMillisecondCounterHiRes() * 0.001 - startTime);**
            addMessageToList (message);
        };

franky47, 2020-08-11:
Unfortunately, it does not. It's possible to use an external time-keeping counter (or base it on `millis`), but since there is no real-time clock on Arduino it might not be precise, and depend on interrupts and the load of the main loop.

jmakurami09, 2020-08-11:
Thank you for your reply.

I am trying to add "MIDI Clock Sync delay" to outgoing messages. Messages in come from a drum machine, but I am trying to add "MIDI Clock Sync delay" like in Ableton live because the drum machine audio is delayed due to a large effects chain so I want the downstream midi to be delayed.

When I introduce "MIDI Clock Sync delay" in Ableton, looking at the MIDI data in MIDI monitor I can see there is a .020s between the incoming and outgoing timestamp and that seems to be the key for making the rest of the MIDI chain "delay".

Could I capture the timestamp parameter from the midi stream , add 20ms to it and send it with the first `send` type that you have in the library?