FortySevenEffects/arduino_midi_library #167: [question] How to use SysExMaxSize

SimonVareille, 2020-08-25:
Hello,
Every time I receive a SysEx message longer than 128 bytes, the message is split into chunks of max 128 bytes. I set SysExMaxSize to something longer than 128 (like 512) but no change occurred. Curiously, if I set SysExMaxSize to something less than 128 (like 64), I still get messages in chunks of max 128 bytes. As if the SysExMaxSize setting didn't change anything.

The Arduino is linked by USB to the computer and I use Hairless-Midiserial to bridge serial to virtual midi ports.
I use python3 and the mido library on computer to send and receive messages.

I can reproduce the issue with an Arduino Mega and an Arduino Uno.

## Minimal example:
Arduino side: *testSysEx.ino*
```C++
#include <MIDI.h>

struct MySettings : public midi::DefaultSettings
{
  static const unsigned SysExMaxSize = 256;
  static const long BaudRate = 115200;
};

MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial, MIDI, MySettings);

void setup()
{ 
  MIDI.begin();
  
  //Disable soft thru so that incoming message are not sent back
  MIDI.turnThruOff();
  //Handle incoming midi messages
  MIDI.setHandleSystemExclusive(systemExclusiveHandler);
}

void loop()
{
  MIDI.read();
}

void systemExclusiveHandler(byte* data, unsigned size) {
  byte data_size[4] = {size>>12, (size>>8)&0xF, (size>>4)&0xF, size&0xF};
  MIDI.sendSysEx(4, data_size); //send the received size
  MIDI.sendSysEx(size-2, data+1); //send back the sysex message (without the first and last bytes)
  if(data[size-1] == 0xF0) //check wether the last byte is 0xF0
    MIDI.sendNoteOn(1, 1, 1);
}
```

Computer side: *testSysEx.py*
```python
# -*- coding: utf-8 -*-

import mido
import time

def message_recv(message):
    if message.type == "sysex":
        print(message.hex())
    else:
        print(message)

print(mido.get_output_names())
print(mido.get_input_names())

outport = mido.open_output("loopMIDI Port in 2", autoreset=True)
inport = mido.open_input("loopMIDI Port out 0", callback=message_recv)

print("first sysex message")
# send 128 bytes
outport.send(mido.Message("sysex", data=[i % 128 for i in range(128-2)]))
time.sleep(1)
print("second sysex message")
# send 138 bytes
outport.send(mido.Message("sysex", data=[i % 128 for i in range(138-2)]))
time.sleep(1)
# close ports
inport.callback = None
outport.close()
inport.close()
```

Expected output:
```
first sysex message
F0 00 00 08 00 F7
F0 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 40 41 42 43 44 45 46 47 48 49 4A 4B 4C 4D 4E 4F 50 51 52 53 54 55 56 57 58 59 5A 5B 5C 5D 5E 5F 60 61 62 63 64 65 66 67 68 69 6A 6B 6C 6D 6E 6F 70 71 72 73 74 75 76 77 78 79 7A 7B 7C 7D F7
second sysex message
F0 00 00 08 0A F7
F0 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 40 41 42 43 44 45 46 47 48 49 4A 4B 4C 4D 4E 4F 50 51 52 53 54 55 56 57 58 59 5A 5B 5C 5D 5E 5F 60 61 62 63 64 65 66 67 68 69 6A 6B 6C 6D 6E 6F 70 71 72 73 74 75 76 77 78 79 7A 7B 7C 7D 7E 7F 00 01 02 03 04 05 06 07 F7
```

Actual output:
```
first sysex message
F0 00 00 08 00 F7
F0 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 40 41 42 43 44 45 46 47 48 49 4A 4B 4C 4D 4E 4F 50 51 52 53 54 55 56 57 58 59 5A 5B 5C 5D 5E 5F 60 61 62 63 64 65 66 67 68 69 6A 6B 6C 6D 6E 6F 70 71 72 73 74 75 76 77 78 79 7A 7B 7C 7D F7
second sysex message
F0 00 00 08 00 F7
F0 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 40 41 42 43 44 45 46 47 48 49 4A 4B 4C 4D 4E 4F 50 51 52 53 54 55 56 57 58 59 5A 5B 5C 5D 5E 5F 60 61 62 63 64 65 66 67 68 69 6A 6B 6C 6D 6E 6F 70 71 72 73 74 75 76 77 78 79 7A 7B 7C 7D F7
note_on channel=0 note=1 velocity=1 time=0
F0 00 00 00 0C F7
F0 7E 7F 00 01 02 03 04 05 06 07 F7
```

The first SysEx message sent by the Arduino encode the size of the received SysEx:
`size = 0xABCD`
SysEx sent:
`F0 0A 0B 0C 0D F7`

The Actual output shows that the long SysEx message is split into two chunks. One of 128 bytes (`0x80`) and one of 12 bytes (`0xC`).

As we can see with `note_on channel=0 note=1 velocity=1 time=0`, the last byte of the first chunk is `0xF0`; as expected (according to line 1001 of *MIDI.hpp*) for long SysEx messages.

Is there anything I did wrong?

kisse66, 2021-06-10:
I have the same problem (<MIDI Library> 5.0.2).  Had to go and modify the library default. My code looks ~identical to above.
I'm using platformIO and mega2560 (working on 4x4 patch bay).
It also seems to require BaudRate defined, otherwise compilation fails:

In file included from .pio\libdeps\megaatmega2560\MIDI Library\src/MIDI.h:35:0,
                 from src\main.cpp:12:
.pio\libdeps\megaatmega2560\MIDI Library\src/serialMIDI.h: In instantiation of 'void midi::SerialMIDI<SerialPort, _Settings>::begin() [with SerialPort = HardwareSerial; _Settings = MyMidiSettings]':
.pio\libdeps\megaatmega2560\MIDI Library\src/MIDI.hpp:75:5:   required from 'void midi::MidiInterface<Transport, _Settings, _Platform>::begin(midi::Channel) [with Transport = midi::SerialMIDI<HardwareSerial, MyMidiSettings>; _Settings = midi::DefaultSettings; _Platform = midi::DefaultPlatform; midi::Channel = unsigned char]'
src\main.cpp:194:42:   required from here
.pio\libdeps\megaatmega2560\MIDI Library\src/serialMIDI.h:62:13: error: 'BaudRate' is not a member of 'midi::SerialMIDI<HardwareSerial, MyMidiSettings>::Settings {aka MyMidiSettings}'
             mSerial.begin(Settings::BaudRate);

kisse66, 2021-06-10:
just tried also Arduino IDE 1.8.13 with board mega2560, similar issue.

```
#include <MIDI.h>

struct MyMidiSettings : public midi::DefaultSettings
{
    static const unsigned SysExMaxSize = 512;
//    static const long BaudRate = 31250;
};

MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial1, MIDI1, MyMidiSettings);

void setup() {
  Serial.begin(115200);
  Serial.println("Started");
  
  MIDI1.begin(MIDI_CHANNEL_OMNI);
}

void loop() {

  if (MIDI1.read()) {
    if (MIDI1.getType() == midi::SystemExclusive) {
      Serial.print(MIDI1.getSysExArrayLength());
      Serial.println(" bytes SYSEX received");
    }
  }

}

```
this gives

_In file included from L:\Arduino\libraries\MIDI_Library\src/MIDI.h:35:0,
                 from L:\Arduino\testMIDI\testMIDI.ino:1:
L:\Arduino\libraries\MIDI_Library\src/serialMIDI.h: In instantiation of 'void midi::SerialMIDI<SerialPort, _Settings>::begin() [with SerialPort = HardwareSerial; _Settings = MyMidiSettings]':
L:\Arduino\libraries\MIDI_Library\src/MIDI.hpp:75:5:   required from 'void midi::MidiInterface<Transport, _Settings, _Platform>::begin(midi::Channel) [with Transport = midi::SerialMIDI<HardwareSerial, MyMidiSettings>; _Settings = midi::DefaultSettings; _Platform = midi::DefaultPlatform; midi::Channel = unsigned char]'
L:\Arduino\testMIDI\testMIDI.ino:18:32:   required from here
L:\Arduino\libraries\MIDI_Library\src/serialMIDI.h:62:13: error: 'BaudRate' is not a member of 'midi::SerialMIDI<HardwareSerial, MyMidiSettings>::Settings {aka MyMidiSettings}'
             mSerial.begin(Settings::BaudRate);
             ^~~~~~~
Using library MIDI_Library at version 5.0.2 in folder: L:\Arduino\libraries\MIDI_Library_ 

Removing comments from the Baudrate makes it compile, but the 512 is not in effect, e.g. receiving 264 byte sysex (Casio CZ) the above sketch prints "33 bytes SYSEX received".  If I go and modify the original header I get correct "264 bytes SYSEX received"

For testing I use UM-ONE USB midi on Windows and F0F7.net web app (reproduced also with MIDI-OX)

Any suggestion to try?

franky47, 2021-06-10:
It looks like the BaudRate issue is related to the fact that it was moved to the Serial transport, @lathoub could you please shine a light on this issue ? It's possible the SysEx size being ignored is related.

lathoub, 2021-06-11:
There are 2 default settings override: the default settings for the Serial transport (`Baudrate`), and the default settings for MIDI (`SysExMaxSize`)

I will test it later, but it should look something like (Leonardo):

```cpp
struct MySerialSettings : public midi::DefaultSerialSettings
{
    static const long BaudRate = 115200;
};

struct MySettings : public midi::DefaultSettings
{
    static const unsigned SysExMaxSize = 256;
};

MIDI_NAMESPACE::SerialMIDI<HardwareSerial, MySerialSettings> serialMIDI(Serial1);
MIDI_NAMESPACE::MidiInterface<MIDI_NAMESPACE::SerialMIDI<HardwareSerial, MySerialSettings>, MySettings> MIDI((MIDI_NAMESPACE::SerialMIDI<HardwareSerial, MySerialSettings>&)serialMIDI);

void setup()
{
  MIDI.begin();
}

void loop()
{
...
```

kisse66, 2021-06-14:
yes, I got it working with

```
#include <MIDI.h>

struct MyMidiSettings : public midi::DefaultSettings
{
    static const unsigned SysExMaxSize = 768;
};

//MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial1, MIDI1, MyMidiSettings);
midi::SerialMIDI<HardwareSerial> serialmidi1(Serial1);
midi::MidiInterface<midi::SerialMIDI<HardwareSerial>, MyMidiSettings> MIDI1(serialmidi1);

void setup() {
  Serial.begin(115200);
  Serial.println("Started");
  
  MIDI1.begin(MIDI_CHANNEL_OMNI);
}

void loop() {

  if (MIDI1.read()) {
    if (MIDI1.getType() == midi::SystemExclusive) {
      Serial.print(MIDI1.getSysExArrayLength());
      Serial.println(" bytes SYSEX received");
    }
  }
}
```

Started
264 bytes SYSEX received

closed 2021-06-19 by SimonVareille