FortySevenEffects/arduino_midi_library #168: [question] Not launching sysex callback

sanotronics, 2020-08-26:
Hi!

I am using this library in this project: https://gitlab.com/yaeltex/ytx-controller/-/tree/unusedConfigUpdate

SysEx handling is done in _IO_PROTOCOL.ino_ and configuration structs are defined in _headers/types.h_

I am sending a sysex configuration to the controller board, each sysex message being a different component to configure, with the BLOCK byte being what kind of component is or if it is a general configuration, and the SECTION bytes being indexes to encoder 1, 2, 3 or button 13 an so on.

I've been using this (almost) flawlessly and I had a good system set up and working, until I decided it'd be a nice idea, 2 weeks before shipping the first units sold for this kind of controllers, to set some extra bytes in the configuration for future implementation of features.

The branch _unusedConfigUpdate_ is the one where I am testing these changes.

Now, I am receiving 2 kind of blocks, the larger ones, Configuration (272 bytes) and Encoders (74 bytes), perfectly, but when the configurator sends the digital or analog blocks (both 47 bytes), the first time through, the sysex callback doesn't seem to be launching.

I have a print right at the beginning of this callback, printing the received message, and I see nothing.

The configurator tool sends 3 times the same message, and as it's not getting an ACK message from the board, it gives up.

If I send the configuration a second time, without changing anything, just hit "Send" a second time, first I see the 3 messages that were sent before as if they were held somewhere, and then I see the new configuration get through complete. So the second time, all the messages arrive perfectly.

I am clueless, I was using until now the 4.4 version of the library with the USB transport, and just to test, I have updated now to 5.0.2, but the same thing happens (I have a different bug with the update sending sysex, some messages go out shorter than they should, but it's something else)

If anything like this ever happened to anyone, I'll apreciate whatever clue you can throw at me.

If you need further info, please let me know.

Thanks!

sanotronics, 2020-08-26:
Ok, I did some testing.

I tried sending the configuration messages using PocketMIDI, just pasting the sysex bytes I want my board to receive as configuration messages.

Like I said, my analog and digital configuration is 47 bytes long, total. 

With PocketMIDI i can edit this messages erasing and adding bytes.

It turns out that if the message is 46, 47 or 48 bytes long, I have to send it 4 times, and only then the sysex callback is launched, yeah, 4 times.

I tried from 30 bytes to 45, all messages launch the callback correctly.

Anything from 49 bytes to 93 bytes, launch the callback correctly as well.

A 94, 95 or 96 byte message, requires 2 messages to launch the callback, 2 times after the second message.

97 to 141 messages work fine, but if until here you found it weird, wait for the message with 142 bytes.

It seems that you have to send 2 messages, and with the second, only one callback is triggered. Then you send one more message and the callback is triggered once, and then with the following message you get 2 callbacks launched.

I am guessing some sort of USB transport bug? 
@franky47 what do you think?

lathoub, 2020-09-19:
Is this the same issue as #170 ?

sanotronics, 2020-09-19:
Hi @lathoub !

I would think they are related, but this one is on reception and https://github.com/FortySevenEffects/arduino_midi_library/issues/170 is when sending a message out.