FortySevenEffects/arduino_midi_library #169: Arduino Nano 33 Low Voltage Signaling

spy590, 2020-08-27:
Greetings! Thanks for the awesome library! I’ve used successfully on basic Uno boards signaling at 5v, but have encountered issues running a very basic MIDI out sample on an Arduino Nano 33 BLE. 

I have adjusted my circuit to use the recommended resistor values for 3.3v signaling from the MIDI specs. In the loop method, I am writing two CC messages with 1000ms delay in between. I have also configured the MIDI instance to use Serial1. 

When connected to Midi Monitor, I seem to be getting numerous random commands including several reset messages. The same code runs fine on an Uno (220 ohm resistors in this case). 

Before I attempt to troubleshoot further, is there any reason this library would not be compatible with the Nano 33?

Thanks!

franky47, 2020-08-31:
Do you have an oscilloscope or any other means to look at the signal outputted by the device ? The MIDI layer of the library has not much to do with the physical transport, other than selecting which port to use. So as long as you used the correct one I think the issue is more an electrical one, rather than a software incompatibility.

dagmanolis, 2021-10-02:
Hello @spy590 , I have successfully used the library on Arduino Nano (even OEM) , but I cannot make it work on an original Arduino nano 33 IoT, any suggestions ? 
 
```
#include <MIDI.h>

MIDI_CREATE_DEFAULT_INSTANCE();

void setup(void)
{
  delay(5000);
  MIDI.begin(MIDI_CHANNEL_OMNI);
  Serial.begin(115200);
}

void loop(void)
{
  MIDI.sendControlChange(1, 10, 1);
  delay(1000);
}
```