FortySevenEffects/arduino_midi_library #170: Not handling usb transport well to send sysex

sanotronics, 2020-08-30:
Hi!

I am sending a 9 byte sysex message with the 5.0.2 version of the library, with an USB instance, and I am not always getting a 9 byte message on the other side.

[image]

With version 4.4 (the fork which used MIDIUSB to include usb transport) all the messages were always succesfully transmitted and I got no shorter messages.

I think maybe the handling of the code index number might be failing?
https://usb.org/sites/default/files/midi10.pdf (page 16)

lathoub, 2020-09-11:
Hi @sanotronics 

Have you tried running the [SysEx example](https://github.com/lathoub/Arduino-USBMIDI/tree/master/examples/SysEx) in the 1.1.2 version of the [arduino USB-MIDI lib](https://github.com/lathoub/Arduino-USBMIDI)? 

I just tried and sending SysEx seems to work normally (MIDI-OX receives SysEx with the correct amount of bytes)

sanotronics, 2020-09-11:
I am using an atsamd21g18 (arduino zero, M0) on the 1.8.12 version.

I haven't tried this example, but I've been using the previous version of the Arduino MIDI lib which had a usb transport support, and the same messages worked alright every time.

I think it has something to do with specific lengths of messages. I can test this example you say with a few more message lengths.

This issue is similar but it also happened on the 4.4 version of the MIDI library, for receiving messages instead of sending them https://github.com/FortySevenEffects/arduino_midi_library/issues/168

lathoub, 2020-09-12:
I have an old ZERO Pro R3 (from the arduino.org days) and no clue how to even get it to run the USB MIDI code or appear in MIDI-OX as a MIDI device.

Try to run the SysEx example code on your M0 and also tell me how you hook it up

lathoub, 2020-09-19:
@sanotronics I tested various sizes of SysEx on an Arduino (over USB) and all passed. I don't have an M0 to test. [Can you test the SysEx example code](https://github.com/lathoub/Arduino-USBMIDI/blob/master/examples/SysEx/SysEx.ino) on your M0 and report back?

sanotronics, 2020-09-19:
Hi! 

I'll give it a go tomorrow!

Will be back with results 💪

sanotronics, 2020-09-21:
Hi!

I tested this, on my board (it's based on an M0 design, but it is a custom board).

Using the v5.0.2 version of this library, I get the following messages on the computer:

This is using Pocket MIDI (MIDI Ox won't receive any messages at all actually, that's weird)
[2020-09-21 16_10_03-Pocket MIDI](https://user-images.githubusercontent.com/4588575/93811009-dd98f580-fc25-11ea-8488-7a460bcb67c4.png)

And this is using a WebMIDI application of our own:
[2020-09-21 16_16_00-Kilowhat - Brave](https://user-images.githubusercontent.com/4588575/93811007-dc67c880-fc25-11ea-8d3b-41353a50ad7e.png)

In both images you can see various lengths of messages received, when I am actually sending the same message over and over again.

lathoub, 2020-09-25:
Please confirm: the above is happening using the SysEx sample code (you are never posting any code in your messages allowing us to reproduce the issue)

sanotronics, 2020-09-25:
I copied and pasted the sysex sample code into my device's firmware.

There is some setup I need to do for it to run, but it is the same code running there in the end.

That's why I didn't copy it here. I will try to upload the exact same example, but as I said before, only switching to the previous version of the library makes it work ok, without any change in the loop code!

Isn't that enough to say that the new version is doing something the previous version wasn't?

lathoub, 2020-09-25:
I tried the sample code on 2 type of boards, all SysEx sizes work fine (MIDI Ox) - trying to get as close to the basic setup as possible.

sanotronics, 2020-09-29:
Ok, I tried just with the SysEx.ino code, and I do get the messages alright on my computer.

I found the reason to the malfunction in my code, which is a timer interrupt every ~400 us. Wrapping the sendSysEx function to disable the timer and reenable it after transmitting, made it work.

I am clueless then, as to why with the previous version of the library I'd get transmission to work properly, and not with the new one :/

If I add the code to handle SysEx reception to the SysEx.ino, this issue is still present: https://github.com/FortySevenEffects/arduino_midi_library/issues/168

Thanks @lathoub, sorry for the misunderstanding, but it really seemed something similar to the issue I linked above.

closed 2020-09-29 by sanotronics