FortySevenEffects/arduino_midi_library #171: CC value incrementing method. Is it possible?

pajczur, 2020-09-08:
Hello,
I would like to ask for support.
I found in your library the method:
`void sendControlChange(DataByte inControlNumber, DataByte inControlValue, Channel inChannel);`

It works great, but it allows only to set absolute value of CC. But I wonder is it possible to set CC value by incrementing it, or of course decrementing. 

I have problem to retreive initial value of parameter from Logic Pro X. If I could retreive that value, then I could start `void sendControlChange()` with that retreived value, and change CC value smoothly.

But now I can't, so when CC value is let's say 127. And in my arduino code I have value 0, then small move with my control (which is encoder) cause CC value jump immedietaly from 127 to 0. I don't like it.

That's why I wonder if there is any chance to incrementing/decrementing CC value instead setting absolute value;

franky47, 2020-09-08:
You should look into RPNs/NRPNs, they have a data increment / decrement feature:
http://www.somascape.org/midi/tech/spec.html#rpns

pajczur, 2020-09-08:
Oh, great thanks for your quick answer. But I am totally greenhorn in Arduino, and in MIDI programing.
And in your link there is no word about arduino. I don't even know how to start, how to use it in Arduino etc.

But OK, I understand in your library there is no function like I ask. OK, I will ask on other forum and make little bit more research.

Thanks a lot,
Best regards

franky47, 2020-09-08:
The library can send RPN and NRPN increment/decrement messages. The link I posted is the MIDI specification, you can get more familiar with what MIDI messages look like there, as for how to use them, check out the Logic manual.

Good luck in your research!

closed 2020-09-08 by franky47