FortySevenEffects/arduino_midi_library #175: Custom MIDI Device name setting?

midiclub604, 2020-09-29:
Hello,

Is it possible to include a setting to change the name of the midi device that shows up on the computer?  I know I can just change the arduino boards.txt, but it would be great to have this option in the midi_Settings.h for example.  The Teensy has this option in the name.c file and it's useful.  Thank you so much for your work on this great project!

lathoub, 2020-10-03:
Hi @midiclub60 , what transport mechanism are you using? Over serial communication, no device name information can be set. However, a name can be set when using MIDI over BLE, USB or rtp. (Teensy connected over USB)

midiclub604, 2020-10-03:
Yes, sorry I meant to specify USB. Currently when the USB-MIDI transport library is loaded onto for example an Arduino Micro, when you plug the arduino board into the computer it will show up as a generic midi device named “Arduino Micro.” If I want to change the name of the device, for example to “My MIDI Controller” I have to go modify the boards.txt file, is there somewhere in the actual USB-MIDI source code I could modify it instead?

lathoub, 2020-10-03:
A quick google search pointed me to: https://github.com/arduino-libraries/MIDIUSB/issues/37
(you would expect it would be simpler, but no)

midiclub604, 2020-10-03:
Yes, I am aware of this method, this is the boards.txt method I was referring to. There must be an alternate way in the USB core source code, I will try and contact the contributors there. Thanks for trying!

closed 2020-10-05 by franky47