FortySevenEffects/arduino_midi_library #176: MIDI Output Only

tomcarlotan, 2020-10-02:
Its not apparent in the resources, but is there a possible setting in the library to just make use of the MIDI output functionality and leave the TX pin free for other uses in the Arduino? Thank you for your work!

franky47, 2020-10-02:
If you meant:
1. Use only the TX pin (MIDI output only, RX is free)
2. Use only the RX pin (MIDI input only, TX is free)

Unfortunately, the serial port is tied both ways on AVR boards. There is a single baud rate (transmission speed) setting for both TX and RX, so unless your other transmission can run at the MIDI rate of 31250 bps, you can't really use the other half of the serial line.

If you can however, it's possible:
- For scenario 1, only use `MIDI.sendXXX` methods to send stuff out, and don't call `MIDI.read`, use `Serial.read` directly to bypass the MIDI parser and read directly from RX.
- For scenario 2, it's the opposite: call `MIDI.read` as usual, don't call any MIDI output methods, and use the serial port directly (`Serial.write`).

In both cases, `MIDI.begin` will setup the whole Serial port for both receiving and sending messages at 31250 bps.

tomcarlotan, 2020-10-02:
My use case is that of 1, I will only use MIDI output functionality and wont use the input. I plan to use RX pin as digital inputs or as an INT pin for the USB host Shield. Will there be conflicts when doing this? Actually I've done this and observed no problems with my project, the USB Host shield works at sending USB-MIDI even if I used the RX pin as the INT pin.

franky47, 2020-10-02:
Are you using a hardware serial output in addition to the USB MIDI ?

I guess you can try and see what happens if you configure your RX pin after the call to `MIDI.begin`, it's possible that the RX circuitry will still try and interpret pin toggles as data bits, but I'm not sure.

tomcarlotan, 2020-10-02:
Yes, Im still using the Hardware Serial along with the USB-HOST as a MIDI Output as well. On top I'm also using the MIDIUSB Library to communicate MIDI via USB to a PC. I observe no problem so far, being able to send on both hardware serial and usb-host, and both send and receive on the built in USB on the ATMEGA32u4 board. I was just wondering if there is a safer way of leaving the RX Pin unused in your library.

franky47, 2020-10-02:
Got it. From what I can read in the AVR user manual, there is a way to clear the RX bit in a register to disable the serial receiver, you might want to check that out before setting the pin up for other tasks. On the 32u4 it looks to be `UCSR1B &= ~(1 << RXEN1)` (see page 222 of the [user guide for that chip](http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-7766-8-bit-AVR-ATmega16U4-32U4_Datasheet.pdf) for the register and bit description).

The order of operations would then be:
```
MIDI.begin(); // on the hardware serial port
UCSR1B &= ~(1 << RXEN1); // clear the RX bit
// setup the RX pin as you like
```

tomcarlotan, 2020-10-02:
Thank you very much! Will look into your recommendation :)

closed 2020-10-02 by franky47

jnettome, 2021-04-10:
Thanks @franky47 ! It help me as well.