FortySevenEffects/arduino_midi_library #178: Sysex problem

ik2wxx, 2020-10-12:
Hi all.
I use this library to send PC and CC command, and they work as expected.
If I try to send Sysex commands, the connected device does not get it.
Sysex command is right, as  I've tried it with different app and midi interface and the device get it.

byte sysexArray[21] = {0xF0, 0x00, 0x20, 0x32, 0x32, 0x2F, 0x63, 0x68, 0x2F, 0x30, 0x31, 0x2F, 0x65, 0x71, 0x2F, 0x6F, 0x6E, 0x20, 0x4F, 0x4E, 0xF7};
MIDI.sendSysEx(sizeof(sysexArray), sysexArray, true);

What's wrong ?

Also noticed that PC and CC work with some devices, not with Behringer devices. 
Any idea ?

Thank you.

franky47, 2020-10-12:
What transport are you using ? DIN cables (hardware serial) or USB ?

ik2wxx, 2020-10-12:
> What transport are you using ? DIN cables (hardware serial) or USB ?

I'm using din cables.
I currently have the midi out only, and waiting for the midi in components.
Btw the other usb midi interface I use (got it from a store) that works with sysex, with only midi out connected to the midi in of the device.

ik2wxx, 2020-10-13:
An update: I switched from Arduino MKR1010 to Arduino Uno and now sysex are working.
What's wrong with this lib on MKR1010 ?
Some midi device get the commands, some other do not.
Any suggestion ?
Tnx

franky47, 2020-10-13:
The only difference I can see is the logic level (3.3V on MKR1010 vs 5V on Uno), what's your circuit for the MIDI out ? Check out [this thread](https://www.midi.org/forum/346-3-3v-vs-5v-and-what-resistors-to-use) for more information. However, that would not explain why only some messages are not received correctly.

Do you have a way to inspect the MIDI traffic sent by your Arduino (ie: MIDI OX or MIDI Monitor using a computer) ?

ik2wxx, 2020-10-13:
> The only difference I can see is the logic level (3.3V on MKR1010 vs 5V on Uno), what's your circuit for the MIDI out ? Check out [this [...]

You're right, I've found that with Arduino Uno, digital output level is 5v, and on MKR 1010 is 3.3v.
Some device work well also with 3.3v, but some do not.
The link you write didn't helped me, but I will search again.
I've debugged the Arduino MKR 1010 midi output and it's 100% ok (connected to a midi/usb adapter to my pc).
Thank you

franky47, 2020-10-13:
Yeah some cheap hardware may omit the MIDI input optocoupler or set it up badly so that it messes up the current loop (which usually makes MIDI voltage-agnostic, when following the specification).

Glad you figured it out, good luck with tweaking the circuit!

closed 2020-10-13 by franky47