FortySevenEffects/arduino_midi_library #180: Incoming message filtering

SandroGrassia, 2020-10-25:
Hi, I'd like to avoid unwanted MIDI messages to be read by my application, I'd like to get rid of them immediately after received; is it possilble?
Thank you!

franky47, 2020-10-25:
I'm not sure what you mean, but if you use [callbacks](https://github.com/FortySevenEffects/arduino_midi_library/wiki/Using-Callbacks), you can "subscribe" only to the message types you're interested in, and not worry about the others.

SandroGrassia, 2020-10-25:
Hi franky47; I got that using callbacks I can filter "types" of messages that i don't need; in my case, I need to go a bit deeper, I would receive in only these Control Change controllers
**1 (Modulation), 7 (Main Volume), 10 (PAN), range 102-120**

and filter all the other Control Change messages.

franky47, 2020-10-25:
In your `handleControlChange` callback, you can test the control number and act on it:

```c++
void handleControlChange(midi::Channel channel, byte control, byte value)
{
    if (control == midi::ModulationWheel)
    {
        // ...
    } 
    else if (control == midi::ChannelVolume)
    {
        // ...
    }
    else if (control >= 102 && control <= 120)
    {
        // ...
    }
    // Otherwise ignore it.
}
```

SandroGrassia, 2020-10-25:
ok, it works perfectly :)
Thank you!

closed 2020-10-25 by franky47