FortySevenEffects/arduino_midi_library #184: mod wheel callback

Pomax, 2020-12-07:
I see there's a pitch bend callback, but there's no mod wheel (cc01) callback. Can one be added?

franky47, 2020-12-08:
You can use the ControlChange callback and check for CC0 there:

```c++
void handleControlChange(byte channel, byte controlNumber, byte value)
{
  if (controlNumber == 0)
  {
    // Handle mod wheel here
  }
}
```

Pomax, 2020-12-08:
While true, it's such an integral part of {pitch, mod, notes} that it seems worth giving it a dedicated callback for folks who don't need "all CC" but do need the mod wheel =)

franky47, 2020-12-08:
If you don't need the rest of the CC, you don't have any more code to write, whereas having a separate case for the mod wheel opens a can of worms: each new callback adds two bytes of RAM, so folks who only need modwheel would not benefit from such a change overall.

Pomax, 2020-12-08:
I see. Could I at least ask that gets added as a code example? Mod wheels are pretty essential, so having the callbacks docs explicitly go "there is no dedicated modwheel callback, because you can already do this with the CC handler in the following way" (or something to that effect) would be super useful documentation.