FortySevenEffects/arduino_midi_library #199: The problem of MIDI library running on NodeMCU board based on ESP8266

Renlong2020, 2021-02-05:
Hello, I am using this MIDI library, this is a very good library.
But now there is a problem
For example, simply send note on in a loop. If running on arduino uno, the serial port baud rate is set to 31250, the serial port debugging software can receive the correct midi signal, but the same code, running on the NodeMCU board based on ESP8266, does not The correct MIDI, is it a problem with my settings or the board?

**Additional context**
Add any other context or screenshots about the feature request here.

franky47, 2021-02-05:
Can you show me your NodeMCU code please ?

Renlong2020, 2021-02-05:
#include <MIDI.h>
//-----------------------------------Midi setting------------------------------------------------------------
struct MySettings : public midi::DefaultSettings
{
    static const unsigned SysExMaxSize = 1024; // Accept SysEx messages up to 1024 bytes long.
    //static const bool UseRunningStatus = true; // My devices seem to be ok with it.
};
//MIDI_CREATE_INSTANCE(HardwareSerial, Serial2, MIDI);
MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial,Serial,MIDI,MySettings);
//-----------------------------------Midi setting------------------------------------------------------------
void
setup()
{
  MIDI.begin(MIDI_CHANNEL_OMNI); //midi
  MIDI.turnThruOff(); //Thru
  }

void loop()
{

MIDI.sendNoteOn(1,127,1);
MIDI.sendNoteOn(2,127,1);
MIDI.sendNoteOn(3,127,1);
delay(1000);
}
NodeMcu :
[image]
arduino:
[image]

franky47, 2021-02-05:
I see nothing in your code or the [ESP8266 Serial documentation](https://arduino-esp8266.readthedocs.io/en/2.7.4_a/reference.html#serial) that would indicate incompatibility. Are you using pin GPIO1 for MIDI output ?

Renlong2020, 2021-02-05:
> 我在您的代码或[ESP8266串行文档中看](https://arduino-esp8266.readthedocs.io/en/2.7.4_a/reference.html#serial)不到任何表明不兼容的信息。您是否将GPIO1引脚用于MIDI输出？

Yes, I use the default serial output, GPIO1 and GPIO3

Renlong2020, 2021-02-05:
> I see nothing in your code or the [ESP8266 Serial documentation](https://arduino-esp8266.readthedocs.io/en/2.7.4_a/reference.html#serial) that would indicate incompatibility. Are you using pin GPIO1 for MIDI output ?

At least the hardware connection is okay. The PC can receive MIDI signals, but the MIDI signals are incorrect. I think the problem may occur in the software, but I don’t know where it is...
Hope to get your help

franky47, 2021-02-05:
What I don't understand is your Arduino screenshot: the serial output does not correspond to the code that sends 3 NoteOn messages (the serial output indicates NoteOn and NoteOff).

Renlong2020, 2021-02-05:
Sorry, I changed the test code after that. The code actually burned to Arduino is the data of the serial portArduino board is no problem. It's a screenshot of 8266. The data received by the serial port is incorrect. For example, the node on should be 90-1-127Actually received 20-01-ff
On 02/05/2021 22:02, François Best wrote:

What I don't understand is your Arduino screenshot: the serial output does not correspond to the code that sends 3 NoteOn messages (the serial output indicates NoteOn and NoteOff).

Renlong2020, 2021-02-06:
Hello, is there any progress on the problem of running simple noteOn on 8266 and not receiving correctly? I studied for a day, but found nothing, I don't know where the problem is.

franky47, 2021-02-06:
I don't have a lot of time to look into this at the moment, also I don't have an ESP8266.

What happens if you send serial messages directly, like:
```cpp
byte counter = 0;

void setup() 
{
  Serial.begin(31250);
}

void setup() 
{
  Serial.write(counter++);
  delay(500);
}
```

underwoodblog, 2021-02-28:
Maby there is something wrong with the terminal program? I use [Protokol ](https://hexler.net/products/protokol) with UART1 on a ESP32 to a Midi-Interface and it works. Resistors for 3V3 and Midi out had to be set to 33 Ohm for 3V3 pullup and 10 Ohm for TX out.

`#define UART1_TXD   GPIO10      /* UART_DEV(1) TxD */
#define UART1_RXD   GPIO9       /* UART_DEV(1) RxD */
MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial1, MIDI, MidiSettings);`

Renlong2020, 2021-03-05:
Finally, the test came out, but the root cause is still unclear. After all, I'm not a hardware major.
The solution is to use IO analog serial port to input and output MIDI signal on esp8266. It can run normally at the baud rate of 31250