FortySevenEffects/arduino_midi_library #200: MIDI.sendAfterTouchPoly(65, 110, 1); not making sense.

MatrixRat, 2021-02-11:
Stripped back to basics, here's code compiled for Teensy LC - 3.6 and Arduino DUE, using IDE 1.8.13 and Teensyduino 1.53.

>#include <MIDI.h>
MIDI_CREATE_INSTANCE(HardwareSerial, Serial1, MIDI);
void setup() {
   MIDI.begin();
   MIDI.turnThruOff();
}
void loop() {
      while (MIDI.read()) {
     // controllers must call .read() to keep the queue clear even if they are not responding to MIDI
  } 
}
void dummy(){
  MIDI.sendAfterTouchPoly(65, 110, 1);
}
>
Compiler returns with:-

> Using library MIDI at version 5.0.2 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\MIDI  'class midi::MidiInterface<midi::SerialMIDI<HardwareSerial> >' has no member named 'sendAfterTouchPoly' >

In all test cases, the IDE shows "sendAfterTouchPoly" highlighted in red however is black for MEGA.

franky47, 2021-02-11:
The correct method is `MIDI.sendAfterTouch`.

With two parameters (value, channel) it is monophonic (acts the same for all notes), and with three parameters (note number, value, channel) it is polyphonic (value per-note).

So your code should be:
```
MIDI.sendAfterTouch(65, 110, 1);
```

MatrixRat, 2021-02-12:
Thank you, Got it working.

closed 2021-02-12 by franky47