FortySevenEffects/arduino_midi_library #204: Simple Synth pitches are incorrect

franky47, 2021-02-22:
See #203.

Correct values of pitches:
https://www.inspiredacoustics.com/en/MIDI_note_numbers_and_center_frequencies

Values must be rounded to the nearest integer, as the `tone` function does not accept floating point values.

akshay4n, 2021-02-23:
@franky47 i would like to work on this

franky47, 2021-02-23:
@akshay4n, awesome, go ahead !

Ping me here if there's anything you need help with.

akshay4n, 2021-02-23:
@franky47  I took a look at #203 and went through [](https://www.inspiredacoustics.com)/en/MIDI_note_numbers_and_center_frequencies and [](https://www.arduino.cc/reference/en/language/functions/advanced-io/tone)/. The problem isn't quite clear to me

franky47, 2021-02-23:
I initially thought that the frequency values were off in the [pitch list](https://github.com/FortySevenEffects/arduino_midi_library/blob/master/examples/SimpleSynth/pitches.h), but the problem is in reading the table.

It starts at B0, which should map to MIDI note number 23. So the index at which we read the pitch should be `noteNumber - 23` to map to the correct note:
https://github.com/FortySevenEffects/arduino_midi_library/blob/master/examples/SimpleSynth/SimpleSynth.ino#L54