FortySevenEffects/arduino_midi_library #210: PC command is not send correctly

Roycinger, 2021-03-21:
- What board are you using ?
Arduino Nano
- What version of the Arduino IDE are you using ?
Sloeber IDE (Eclipse)
- How are you using MIDI ?
    - [x ]  Hardware Serial (DIN plugs)
    - [ ]  USB
    - [ ]  Other (please specify)
- Is your problem related to:
    - [ ]  MIDI Input (reading messages from other devices)
    - [x ]  MIDI Output (sending messages to other devices)
- How comfortable are you with code ?
    - [ ]  Complete beginner
    - [ ]  I've done basic projects
    - [x ]  I know my way around C/C++
    - [ ]  Advanced / professional

## Describe your project and what you expect to happen:
When a button is pressed, a PC should be sent to another device.
I expect the device (Line 6 HX Stomp) to switch to the provided preset

## Describe your problem (what does not work):
I use the command as follows:
[image]
I experience 2 problems here:
1) The HX switches to a non-adressed preset (PC060 instead of PC019in this case).
2) If I press the switch multiple times, it toggles between 2 Presets (PC000 and PC060) when it should only switch to one preset, not toggle between the both.

I do have a Serial to USB cable incoming to monitor exactly what's going on on the PC. In the meantime I wanted to see if someone could have an idea if I'm using the command not as intended.

## Steps to reproduce

I can upload the code to pastebin if required

franky47, 2021-03-21:
The code seems fine, according to what's output on the debug console.

Waiting to see what the MIDI monitor says. In the mean time, can you post more of your code that goes around this block ?

Roycinger, 2021-03-21:
> The code seems fine, according to what's output on the debug console.
> 
> Waiting to see what the MIDI monitor says. In the mean time, can you post more of your code that goes around this block ?

Sure, here's the pastebin of the .ino, header and cpp file (that I implement as a lib):
https://pastebin.com/bTuyV1x4

franky47, 2021-03-21:
I see the problem: you use the Serial port both for MIDI and for serial debugging.

In `setup`, the call to `Serial.begin(9600)` resets the band rate for the MIDI port, and your device has a hard time interpreting your debug commands as MIDI (wrong byte stream at the wrong speed).

Unfortunately, on single-UART chips it's hard to debug and send MIDI at the same time. You could try using a SoftwareSerial port for development, then turn off serial debugging and switch back to hardware serial for "production". Check out the examples to see how to do that.

Roycinger, 2021-03-21:
> I see the problem: you use the Serial port both for MIDI and for serial debugging.
> 
> In `setup`, the call to `Serial.begin(9600)` resets the baud rate for the MIDI port, and your device has a hard time interpreting your debug commands as MIDI (wrong byte stream at the wrong speed). This also explains the weird characters at the beginning of the line in the serial monitor: those two bytes are the actual Program Change interpreted as ASCII.
> 
> Unfortunately, on single-UART chips it's hard to debug and send MIDI at the same time. You could try using a SoftwareSerial port for development, then turn off serial debugging and switch back to hardware serial for "production". Check out the examples to see how to do that.

That makes sense and solved the issue, thanks a lot!
Should I just change the baud rate for the serial monitor?
I'll look for the examples you mentioned to accomplish this.

franky47, 2021-03-21:
You could use 115200 for the serial monitor, which is faster and won't impact timings as much as 9600. MIDI uses a baud rate of 31250.

Roycinger, 2021-03-21:
Thanks for helping out franky, I'll close this issue as I consider it solved :)

closed 2021-03-21 by Roycinger