FortySevenEffects/arduino_midi_library #211: Enable inversion of output messages

modularev, 2021-03-25:
**Is your feature request related to a problem? Please describe.**
Connecting two boards directly with RX/TX pins and sending MIDI messages back and forth doesn't work.

**Describe the solution you'd like**
Enable inversion of the messages wich are sent by the TX output.

**Describe alternatives you've considered**
Put a hardware inverter between the devices.

**Additional context**
-

ht93, 2021-12-16:
Hi @franky47, is it possible to add inversion to RX and TX pin (like a bool in setting that can individually invert the RX or TX)? Thanks a lot!

franky47, 2021-12-16:
Just so that I understand, we're talking about bit flipping of the data coming out of TX / in from RX here, right ?

Like 10010000 (NoteOn on channel 1) would become 01101111.

This could be a transport option. cc @lathoub

ht93, 2021-12-16:
@franky47 thank you very much for the reply. 

Your description is correct. The signal that RX going to get is default 0 and all message have flipped.

For the actual use case, I am not doing the same thing as mentioned by the author. My use case is that, I am currently designing a project that use trs cable for midi input, and I want to only use the tip and sink to carry the Current Sink and GND of midi, without the Current Source. The Current Sink (thru optocoupler) will connected to RX pin. So the RX input is inverted.

franky47, 2021-12-16:
I see. Hardware-wise, a word of advice: this kind of setup may be susceptible to ground loop noise which may cause issues with the correct reception of messages. This is why MIDI has a current loop and optocouplers: to shield each device from each other's ground (they usually are grounded at the audio and power level already). You should probably use the current sink and source rather than the GND, which should be left unconnected at the MIDI input, and then your bit polarity issue disappears.

franky47, 2021-12-16:
@modularev what was your hardware setup in which you couldn't have two boards communicating directly by TX -> RX ?

ht93, 2021-12-17:
@franky47 yeah, I have fully consider that, I will use a trrs connector. The tip is going to reserve for other function. First ring is for current sink and link to positive input of optocoupler. Second ring is connect to GND of midi input and negative optocoupler. Trrs sink is connect to GND. If user use normal trs plug, it will create ground loop. But if user use trrs plug, it will not create ground loop. With this setup, I can use tip for other thing.