FortySevenEffects/arduino_midi_library #215: Serial Midi with custom BaudRate and Library version 5

ClaudioCas, 2021-04-18:
Dear all,

I'm using the following code for initialize my Midi over Usb Serial with custom boudrate according with this doumentation (https://github.com/FortySevenEffects/arduino_midi_library/wiki/Using-custom-Settings) to use Arduino Mega and Hairless MIDI.
```

  #include <MIDI.h>
  struct CustomBaudRate : public midi::DefaultSettings{

      static const long BaudRate = 19200;

  };
  MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial, MIDI, CustomBaudRate);
```
It works great with versione 4.3.1.

I upgrade the library to version 5.0.2 for use native HID with Leonardo. The Usb Midi Native version works relly well, but now if I try to compile the old code (for Usb Serial) with new library I had the following errors:

```
In file included from C:\Users\ClaudioCas\Documents\Arduino\libraries\MIDI_Library\src/MIDI.h:35:0,
                 from c:\Users\ClaudioCas\Documents\Arduino\MidiProject\MidiProject.ino:27:
C:\Users\ClaudioCas\Documents\Arduino\libraries\MIDI_Library\src/serialMIDI.h:122:71: error: no matching function for call to 'midi::SerialMIDI<HardwareSerial, CustomBaudRate>::SerialMIDI(Serial_&)'
     MIDI_NAMESPACE::SerialMIDI<Type, Settings> serial##Name(SerialPort);\
                                                                       ^
c:\Users\ClaudioCas\Documents\Arduino\MidiProject\MidiProject.ino:36:3: note: in expansion of macro 'MIDI_CREATE_CUSTOM_INSTANCE'
   MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial, MIDI, CustomBaudRate);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\ClaudioCas\Documents\Arduino\libraries\MIDI_Library\src/serialMIDI.h:48:2: note: candidate: midi::SerialMIDI<SerialPort, _Settings>::SerialMIDI(SerialPort&) [with SerialPort = HardwareSerial; _Settings = CustomBaudRate]
  SerialMIDI(SerialPort& inSerial)
  ^~~~~~~~~~
C:\Users\ClaudioCas\Documents\Arduino\libraries\MIDI_Library\src/serialMIDI.h:48:2: note:   no known conversion for argument 1 from 'Serial_' to 'HardwareSerial&'
C:\Users\ClaudioCas\Documents\Arduino\libraries\MIDI_Library\src/serialMIDI.h:43:7: note: candidate: constexpr midi::SerialMIDI<HardwareSerial, CustomBaudRate>::SerialMIDI(const midi::SerialMIDI<HardwareSerial, CustomBaudRate>&)
 class SerialMIDI
       ^~~~~~~~~~
C:\Users\ClaudioCas\Documents\Arduino\libraries\MIDI_Library\src/serialMIDI.h:43:7: note:   no known conversion for argument 1 from 'Serial_' to 'const midi::SerialMIDI<HardwareSerial, CustomBaudRate>&'
C:\Users\ClaudioCas\Documents\Arduino\libraries\MIDI_Library\src/serialMIDI.h:43:7: note: candidate: constexpr midi::SerialMIDI<HardwareSerial, CustomBaudRate>::SerialMIDI(midi::SerialMIDI<HardwareSerial, CustomBaudRate>&&)
C:\Users\ClaudioCas\Documents\Arduino\libraries\MIDI_Library\src/serialMIDI.h:43:7: note:   no known conversion for argument 1 from 'Serial_' to 'midi::SerialMIDI<HardwareSerial, CustomBaudRate>&&'
Compilation error: Error: 2 UNKNOWN: exit status 1
```
I'm using Arduino IDE 2.0.0 beta 5

Thanks a lot
Claudio

franky47, 2021-04-18:
I cannot reproduce your error, your example code compiles fine on both Arduino IDE 1.8 and 2.0.

ClaudioCas, 2021-04-18:
Dear Franky,
first of all, thanks a lot for your kind and quick reply.
[image]

This is a empty sketch and doesn't copile for me. I double ckeck, I have 5.0.2 Midi Library

```

  #include <MIDI.h>

  //Arduno Mega + Hairless Midi = 19200;
  //Tips: porta del pc dentro gestione periferiche, hailess midi e arduino devono avere la stessa velocità
  struct CustomBaudRate : public midi::DefaultSettings{

      static const long BaudRate = 19200;

  };
  MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial, MIDI, CustomBaudRate);

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

```

lathoub, 2021-05-13:
Leonardo uses `Serial1`, not `Serial `in the above example.
Change to:
`MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial1, MIDI, CustomBaudRate);`

See also:
https://github.com/FortySevenEffects/arduino_midi_library/blob/d9149d19df867abbf78cfcd37ef9d0e14dedca7f/src/serialMIDI.h#L105-L109

steeley, 2022-04-02:
I have the same issue. Custom settings does not work at all. I tried the example above for Leonardo and still get the error:
`MIDI_Library/src/serialMIDI.h:62:13: error: 'BaudRate' is not a member of 'midi::SerialMIDI<HardwareSerial, MySettings>::Settings {aka MySettings}'
             mSerial.begin(Settings::BaudRate);
             ^~~~~~~
exit status 1
Error compiling for board Arduino Leonardo.
`

copych, 2024-03-09:
MIDI Library ver.5.0.2
ESP core 2.0.14
target: ESP32S3 (LOLIN S3 Pro)
I can compile in Arduino IDE 1.8.20 , but cannot in Arduino IDE 2.3.1
target: ESP32S3 (LOLIN S3 Pro)
the errors start with
error: no matching function for call to 'midi::SerialMIDI<HardwareSerial, CustomBaudRateSettings>::SerialMIDI(HWCDC&)'
 MIDI_NAMESPACE::SerialMIDI<HardwareSerial, CustomBaudRateSettings> serialMIDI(Serial);

lathoub, 2024-03-11:
When using the custom settings, make sure you use either a clone or copy from this repo, not a Release version (eg from the Arduino lib manager).

A new release is upcoming

01GOD, 2024-10-24:
> MIDI Library ver.5.0.2 ESP core 2.0.14 target: ESP32S3 (LOLIN S3 Pro) I can compile in Arduino IDE 1.8.20 , but cannot in Arduino IDE 2.3.1 target: ESP32S3 (LOLIN S3 Pro) the errors start with
>
> ``` [...]

Saw now while looking for a solution while compiling that acidbox project. Found a solution?