FortySevenEffects/arduino_midi_library #259: Add Bank Select LSB enum(0x20) to MidiControlChangeNumber

muxa, 2021-11-27:
**Is your feature request related to a problem? Please describe.**
I want to use MidiControlChangeNumber enum to detect controller changes, and one of the controllers I'm missing is Bank Change LSB (`0x20`).

**Describe the solution you'd like**
Add `BankSelectLSB = 32` to `MidiControlChangeNumber` enum.

**Describe alternatives you've considered**
I am using a literal value `0x20` but it would be nice to have consistency. 

**Additional context**
N/A

franky47, 2021-11-27:
That would be a great addition indeed, along with all the other missing definitions:
http://www.somascape.org/midi/tech/spec.html#ctrlnums

Would you like to open a Pull Request ?

franky47, 2021-11-27:
Note: a function to send high resolution controllers would also be a great addition. It would handle:
- Mapping the input data range (an `unsigned int` with 14 bits of precision, between 0 and 16383) into the two 7-bit data bytes
- Send the MSB first, then the LSB

Example API:
```c++
const unsigned int value = 8192; // half way
const byte channel = 1;
MIDI.sendHighResolutionControlChange(midi::ModulationWheel, value, channel);
```

muxa, 2021-11-28:
I'll open a PR to adding new definitions. 

I don't have a test rig to test MIDI out yet, so your other suggestion is out of scope for now 😄

franky47, 2021-11-28:
You can add a unit test here:
https://github.com/FortySevenEffects/arduino_midi_library/blob/master/test/unit-tests/tests/unit-tests_MidiOutput.cpp

closed 2021-11-30 by muxa