FortySevenEffects/arduino_midi_library #287: MIDI Library with Arduino Uno WiFi Rev2 not working

labamichnetvoll, 2022-06-11:
I tried to use the Arduino Uno WiFi Rev2 with this MIDI lIbrary and I dont get it to run.

Its a very simple Output Code:
`#include <MIDI.h>

#define LED LED_BUILTIN

MIDI_CREATE_DEFAULT_INSTANCE();

void setup()
{
  pinMode(LED, OUTPUT);
  MIDI.begin(MIDI_CHANNEL_OFF);
}

void loop()
{
  digitalWrite(LED, HIGH);
  MIDI.sendNoteOn(60, 127, 1);
  delay(1000);		       
  MIDI.sendNoteOff(60, 0, 1);
  digitalWrite(LED, LOW);
  delay(1000);
}
`

The same code on the  normal Arduino Uno runs fine, so the issue isn't on my hardware. It looks like the problem comes from the Arduino Uno WiFi Rev2. 
Any suggestions for me?

Thanks in advance

lathoub, 2022-06-11:
What does not run? No led blinking? Or not receiving MIDI on the other side? (Not sure if the MIDI_CHANNEL_OFF inhibits sending notes)

labamichnetvoll, 2022-06-11:
On a normal arduino uno this code works fine. So I don’t think it’s a software problem except for maybe another instance of the midi lib.

franky47, 2022-06-11:
What is it you expect it to do exactly?

Theoretically, this sketch should:
- Turn on an LED connected to pin `LED_BUILTIN` (the LED on the board)
- Send a NoteOn MIDI message to the default hardware serial port (TX on D1)
- Wait one second
- Send a NoteOff MIDI message to the default hardware serial port (TX on D1)
- Turn the LED off
- Wait one second, and repeat

If you're looking into doing more advanced MIDI routing (like BLE or over IP/WiFi), you'll need to look into [Transports](https://github.com/FortySevenEffects/arduino_midi_library#other-transport-mechanisms).

labamichnetvoll, 2022-06-11:
Yes. There I agree with your explanation of the theoretical result. 

In Reality: 
The led is blinking fine but no MIDI Messages receive my computer at the program “Pocket MIDI”. I’ve tried the exact same code on an other arduino uno board (not the WiFi Rev2) and there with the same configuration and setup and hardware the code works fine.
So blinking of the led and sending a MIDI Message every second.

but for some reason the uno wifi rev2 has some problems with this code.

labamichnetvoll, 2022-06-12:
I've found the issue:

The Arduino Uno WiFi Rev2 has his first Hardwareserial called "Serial" only connected to the USB Port.
The HardwarePins TX and RX are connected to "Serial1", so it not possible with this arduino to call the `MIDI_CREATE_DEFAULT_INSTANCE` function. instead you have to use `MIDI_CREATE_INSTANCE(HardwareSerial,` Serial1, MIDI);`

closed 2022-06-12 by labamichnetvoll

franky47, 2022-06-12:
Thanks for the feedback.

We have a way to detect the default hardware serial port name, it might need an update to be compatible with the Uno WiFi Rev2.