FortySevenEffects/arduino_midi_library #290: No data sent with ESP32

highway11, 2022-07-04:
Has anyone successfully used this library with an ESP32 and Hairless Midi Serial Bridge (Windows)?
I have a very simple program to test, but it seems to be sending no data via the serial connection:

`#include <MIDI.h>

// Create and bind the MIDI interface to the default hardware Serial port
//MIDI_CREATE_DEFAULT_INSTANCE();
MIDI_CREATE_INSTANCE(HardwareSerial, Serial, MIDI);

void setup() {
  MIDI.begin(MIDI_CHANNEL_OMNI);  // Listen to all incoming messages
  Serial.begin(115200);
}

void loop() {
  
    MIDI.sendNoteOn(1, 127, 1);
    //MIDI.read();
    delay(1000);
}`

Is there something I need to be doing differently? Monitoring the serial port shows no data is being sent

lathoub, 2022-07-04:
Hi @highway11 
I use this lib with an ESP32, so it can work :-)

What I recommend:

1. `Serial` will not work (`serial` is used for programming the ESP32), I use `Serial2`
2. ESP32 uses 3.3V, not 5V (other resistor values, do a google search) 
3. use the [customBaudRate](https://github.com/FortySevenEffects/arduino_midi_library/blob/master/examples/CustomBaudRate/CustomBaudRate.ino) example

Good luck!

highway11, 2022-07-04:
Thanks for the info @lathoub 

I'm trying to send midi data over the same USB connection I use to program the ESP32, using the Hairless Midi Serial Bridge.
https://projectgus.github.io/hairless-midiserial/

1.When you say use Serial2, does that mean you are sending midi data out of a different connection then the standard USB on the ESP32?
2. I don't think voltage is an issue unless I'm using different pins/connection than the standard USB port?

lathoub, 2022-07-04:
Totally confused. An ESP32 does not have a USB port (or are you talking about a USB in from of a ESP32 (eg DOIT ESP32 Devkit v1.0) that is used program it - that will not support MIDI)

highway11, 2022-07-04:
Yes, its a ESP32 Devkit sorry for the confusion. I was using RTPMidi to send midi commands via wifi, but I was hoping to switch to USB for it to be more robust connection. My idea was just plug in the ESP32 Devkit into my PC and have it send midi messages through the serial connection which Hairless Midi Serial Bridge is supposed to interpret and send out to another midi device

highway11, 2022-07-05:
Is there a reason it can't send MIDI commands over the same Serial/USB port I use to program it? I thought that was the purpose of the Hairless Midi Serial Bridge software, to monitor that serial port and redirect MIDI commands? 

Failing that, is there another way you suggest connecting my ESP32 Devkit to a windows PC to send MIDI commands via USB? Do I need to create a 2nd USB connection from some other pins on the ESP32?

lathoub, 2022-07-05:
Got it, unconfused now :-) I never tried it that way. Do you have a serial sniffer on your computers serial port? What does that show?

highway11, 2022-07-05:
That's the strange thing, the serial monitor shows no output whatsoever. I thought the `MIDI.sendNoteOn(1, 127, 1);` command should send some data through serial. 

I also tested with an older Midi library (ardumidi) and it sends data through serial but it seems that the Hairless Midi Bridge doesn't know how to interpret that data, so I'm trying the more recommended library which is this one