FortySevenEffects/arduino_midi_library #300: ATMEGA4809 (Arduino Nano "Every" board) should use Serial1 not Serial

warmans, 2022-09-18:
## Context

Please answer a few questions to help us understand your problem better and guide you to a solution:

- What board are you using ?
    - `Arduino Nano Every Board (ATMEGA4809)`
- What version of the Arduino IDE are you using ?
    - `2.0`
- How are you using MIDI ?
    - [x]  Hardware Serial (DIN plugs)
    - [ ]  USB
    - [ ]  Other (please specify)
- Is your problem related to:
    - [ ]  MIDI Input (reading messages from other devices)
    - [x]  MIDI Output (sending messages to other devices)
- How comfortable are you with code ?
    - [ ]  Complete beginner
    - [x]  I've done basic projects
    - [ ]  I know my way around C/C++
    - [ ]  Advanced / professional

## Describe your project and what you expect to happen:

I was using an older third party Arduino Nano for development and at the end swapped in a newer ATMEGA4809 board (https://docs.arduino.cc/hardware/nano-every). After switching I wasn't getting any output. Turns out the newer Nano boards use Serial1 by default. I was able to fix this using `MIDI_CREATE_INSTANCE` instead of `MIDI_CREATE_DEFAULT_INSTANCE` but I think the library does switch between Serial and Serial1 automatically for other boards. 

Not really a bug, but I guess the library should take this new Nano Every board into account with `CREATE_DEFAULT_INSTANCE`

franky47, 2022-09-18:
Thanks for reporting, the offending line is here:
https://github.com/FortySevenEffects/arduino_midi_library/blob/master/src/serialMIDI.h#L110

Could you try and find the definition for that board and report back here please? So I could update the automatic serial port selection.

Thanks!

warmans, 2022-09-18:
To be honest I'm not sure how to get the name. But `ARDUINO_AVR_NANO_EVERY` seems to be correct based on what I've found online.

aequi42, 2022-11-07:
Hi, I've had issues getting my Board to run and saw this issue.
The workaround used by @warmans worked for me as well.

The Board name is indeed `ARDUINO_AVR_NANO_EVERY`. The full definition, found in `~/.arduino15/packages/arduino/hardware/megaavr/1.8.7/boards.txt` is:

```

nona4809.name=Arduino Nano Every

nona4809.vid.0=0x2341
nona4809.pid.0=0x0058

nona4809.upload.tool=avrdude
nona4809.upload.protocol=jtag2updi
nona4809.upload.maximum_size=49152
nona4809.upload.maximum_data_size=6144
nona4809.upload.speed=115200
nona4809.upload.use_1200bps_touch=true
nona4809.upload.extra_params=-P{serial.port}

nona4809.build.mcu=atmega4809
nona4809.build.f_cpu=16000000L
nona4809.build.board=AVR_NANO_EVERY
nona4809.build.core=arduino
nona4809.build.variant=nona4809
nona4809.build.text_section_start=.text=0x0
nona4809.build.extra_flags={build.328emulation} -DMILLIS_USE_TIMERB3 -DNO_EXTERNAL_I2C_PULLUP
#nona4809.build.extra_flags=-B{runtime.tools.atpack.path}/gcc/dev/{build.mcu}

nona4809.bootloader.tool=avrdude
nona4809.bootloader.file=atmega4809_uart_bl.hex
nona4809.bootloader.SYSCFG0=0xC9
nona4809.bootloader.BOOTEND=0x00
nona4809.bootloader.OSCCFG=0x01
nona4809.fuses.file=fuses_4809.bin

menu.mode=Registers emulation
nona4809.menu.mode.on=ATMEGA328
nona4809.menu.mode.on.build.328emulation=-DAVR_NANO_4809_328MODE
nona4809.menu.mode.off=None (ATMEGA4809)
nona4809.menu.mode.off.build.328emulation=

```

(Also mentioned [here](https://arduino.stackexchange.com/a/85564))

I couldn't find any specification for `Serial1` other than this sentence in [this Guide](https://www.arduino.cc/en/Guide/NANOEvery):
> The interface to the SAMD11 / USB is Serial and the the one available on pins RX0 and TX1 is Serial1.