FortySevenEffects/arduino_midi_library #308: MIDI not working with ESP32 S3 with Serial2

mastrumisciu, 2022-11-18:
Hi I'm using both an ESP32 WROOM 32 and a ESP32-S3-DevKitC-1 for some MIDI projects with Arduino.
The code I'm using is the one below and for both board I'm using the following wiring to the 5 din connector and using resistors for 3,3volt.
https://eu-central-1-02860049-view.menlosecurity.com/c/0/i/aHR0cHM6Ly93d3cubWlkaS5vcmcvaW1hZ2VzL3NwZWNzL1NjcmVlbl9TaG90XzIwMjAtMDctMThfYXRfMTIuMjAuMDhfUE0ucG5n

By using pin 17 as Tx I get the code working on the WROOM board only, I can upload the code with no errors on the S3 board as well but no MIDI message are sent from pin 17.
Is there anyone using the S3 for MIDI over Serial2 that can help me on `this?
Thanks!

``

#include <MIDI.h>
#include <HardwareSerial.h>
#include <esp_now.h>

struct Serial2MIDISettings : public midi::DefaultSettings
{
static const long BaudRate = 31250;
static const int8_t TxPin = 17;
};

MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial2, DIN_MIDI, Serial2MIDISettings);

void setup() {
Serial.begin(9600);
DIN_MIDI.begin(MIDI_CHANNEL_OMNI);
delay(300);
DIN_MIDI.setHandleProgramChange(handleProgramChange);
}

void loop() {
DIN_MIDI.sendProgramChange(42, 10);
delay(100);
}

``
- What board are you using ?
    - ESP32wroom and ESP32 S3 Devkit
    - 
- What version of the Arduino IDE are you using ?
    - 1.8.5
    - 
- How are you using MIDI ?
    - DIN plug
    - 
- Is your problem related to:
    - MIDI Output (sending messages to other devices)
    -
- How comfortable are you with code ?
    - I've done basic projects

## Describe your project and what you expect to happen:

## Steps to reproduce

mastrumisciu, 2022-11-18:
I tried to simplify the code, see below, but again it works perfectly on the WROOM board and there's no message coming from the S3

``
#include <MIDI.h>
#include <HardwareSerial.h>
#include <esp_now.h>

MIDI_CREATE_INSTANCE(HardwareSerial, Serial2, midi2)

void setup() {
  Serial2.begin(31250);
  midi2.begin(MIDI_CHANNEL_OMNI);
}

void loop() {
  midi2.sendProgramChange(42, 10);
  delay(100);
  Serial.println("LOOP");
}
``

mastrumisciu, 2022-11-19:
Using Serial 1 and related pin 16 the code works on S3 board as well, I checked if the pin 17 of the dev kit board is connected to the esp3 s3 module pin and I confirm it is so I think there’s some bug in Serial2, Thanks for your help

cyberdevnet, 2023-01-19:
I have a similar problem where I can only send and not receive, but if invert the wiring then I can receive but not send. Using Serial2 (RXD2 16, TXD2 17)

RaphaelDives, 2024-10-28:
Hey,
I'm working on a Lolin Mini S2 and had a similar problem.
Using "Serial1" would compile perfectly, but this not supported, because of free assignment of pins, or whatever.
However, specifying the Pins worked or me.

e.g. Serial1.begin(31250, SERIAL_8N1, 11, 12);

Maybe this could be helpful?
Cheers
Raphael