FortySevenEffects/arduino_midi_library #310: DIN MIDI AND BLE MIDI use at the same time

vamix2012, 2022-11-23:
### Discussed in https://github.com/FortySevenEffects/arduino_midi_library/discussions/309

<div type='discussions-op-text'>

<sup>Originally posted by **vamix2012** November 23, 2022</sup>
I use the MIDI Library with my midi controller and I am using esp32, I tried hard to achieve 
DIN MIDI OUT and BLE MIDI OUT at the same time, 
One is working and the second one is not, can you guide me please maybe I understood something wrong, is that even possible?

I tried to use this 

```
MIDI_CREATE_INSTANCE(HardwareSerial, Serial1, SERIAL_MIDI_OUTPUT);
BLEMIDI_CREATE_INSTANCE("Midi Fun", MIDI);
```

on the setup....
```
  MIDI.begin(MIDI_CHANNEL_OMNI);
  SERIAL_MIDI_OUTPUT.begin(MIDI_CHANNEL_OMNI);
```

and then used it in the code like this 

```
    MIDI.sendControlChange(CC_Expression, expression, 3);
SERIAL_MIDI_OUTPUT.sendControlChange(CC_Expression, expression, 3);
```

The BLE works well but the serial midi is not outputting anything

can you please help me with that? maybe is something simple, I am stuck on this.

Thanks in advance
Cristian</div>

franky47, 2022-11-23:
Maybe look into https://github.com/FortySevenEffects/arduino_midi_library/issues/290 ?  Looks like you may need to use Serial2.

vamix2012, 2022-11-24:
I've tried to use the Serial2 but it still does not work, the BLE is working fine while I get no output from the Serial Port, I have to mention that I tried also reassigning the TX pin but that didn't work as well, do you think is there some way to set the Serial Midi out on it's original TX 

Like MIDI_CREATE_DEFAULT_INSTANCE();

And configure the ble instance on other way?

 I have to mention that I use the wrover-e module which does not have the pins for rx1,tx1 and rx2,tx2 exposed....

closed 2022-11-24 by vamix2012

vamix2012, 2022-11-24:
I inverted the two lines where the midi instances are initialized as following:

MIDI_CREATE_DEFAULT_INSTANCE();
BLE_CREATE_INSTANCE("NAME", BLMIDI);

THEN

MIDI.begin();
BLMIDI.begin();

It works like this when serial midi is default initialized and the BLE or USB are initialized with something other than MIDI.

Thanks for the tips, helped me to look at the problem from other perspective.

Best Regards 
Cristian