FortySevenEffects/arduino_midi_library #330: Unable to override one of the default settings

amowry, 2023-07-14:
I started this as a discussion but I'm adding it as an issue because as far as I can tell I'm following the instructions (though I certainly could be wrong about that :)). I'm trying to override one of the default settings (1-byte parsing):

```c++
struct MySettings : public MIDI_NAMESPACE::DefaultSettings
 {
        static const bool Use1ByteParsing = false;  //parse more than 1 byte per MIDI.read()
 };

// Create instances of the Arduino MIDI Library.
MIDI_CREATE_CUSTOM_INSTANCE(BLEMidi, blemidi, BLEMIDI, MySettings);
MIDI_CREATE_CUSTOM_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI, MySettings);

// Create instances of the Arduino MIDI Library.
//MIDI_CREATE_INSTANCE(BLEMidi, blemidi, BLEMIDI);
//MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);
```

The default instances (commented out above) work fine, but the custom instances result in errors concerning the baud rate:

```
In file included from c:\Users\Andrew\Documents\Arduino\libraries\MIDI_Library\src/MIDI.h:35,
                 from C:\Users\Andrew\Documents\GitHub\WARBL2\warbl2_firmware\warbl2_firmware.ino:12:
c:\Users\Andrew\Documents\Arduino\libraries\MIDI_Library\src/serialMIDI.h: In instantiation of 'void midi::SerialMIDI<SerialPort, _Settings>::begin() [with SerialPort = Adafruit_USBD_MIDI; _Settings = MySettings]':
c:\Users\Andrew\Documents\Arduino\libraries\MIDI_Library\src/MIDI.hpp:75:5:   required from 'void midi::MidiInterface<Transport, _Settings, _Platform>::begin(midi::Channel) [with Transport = midi::SerialMIDI<Adafruit_USBD_MIDI, MySettings>; _Settings = midi::DefaultSettings; _Platform = midi::DefaultPlatform; midi::Channel = unsigned char]'
C:\Users\Andrew\Documents\GitHub\WARBL2\warbl2_firmware\warbl2_firmware.ino:596:33:   required from here
c:\Users\Andrew\Documents\Arduino\libraries\MIDI_Library\src/serialMIDI.h:62:13: error: 'BaudRate' is not a member of 'midi::SerialMIDI<Adafruit_USBD_MIDI, MySettings>::Settings' {aka 'MySettings'}
   62 |             mSerial.begin(Settings::BaudRate);
      |             ^~~~~~~
c:\Users\Andrew\Documents\Arduino\libraries\MIDI_Library\src/serialMIDI.h: In instantiation of 'void midi::SerialMIDI<SerialPort, _Settings>::begin() [with SerialPort = BLEMidi; _Settings = MySettings]':
c:\Users\Andrew\Documents\Arduino\libraries\MIDI_Library\src/MIDI.hpp:75:5:   required from 'void midi::MidiInterface<Transport, _Settings, _Platform>::begin(midi::Channel) [with Transport = midi::SerialMIDI<BLEMidi, MySettings>; _Settings = midi::DefaultSettings; _Platform = midi::DefaultPlatform; midi::Channel = unsigned char]'
C:\Users\Andrew\Documents\GitHub\WARBL2\warbl2_firmware\warbl2_firmware.ino:623:36:   required from here
c:\Users\Andrew\Documents\Arduino\libraries\MIDI_Library\src/serialMIDI.h:62:13: error: 'BaudRate' is not a member of 'midi::SerialMIDI<BLEMidi, MySettings>::Settings' {aka 'MySettings'}
```

Could someone tell me what I'm doing wrong? Thanks!

_Originally posted by @amowry in https://github.com/FortySevenEffects/arduino_midi_library/discussions/329_

amowry, 2023-07-14:
These are instances of BLE MIDI and USB MIDI--I'm using modifications of the Adafruit Bluefruit and TinyUSB MIDI examples below. I'm not sure if those are implemented as serial in the Adafruit libraries?:

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/Peripheral/blemidi/blemidi.ino

https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/examples/MIDI/midi_test/midi_test.ino

It looks like it's the same baud rate error that was occurring here: https://github.com/FortySevenEffects/arduino_midi_library/discussions/323

lathoub, 2023-07-14:
What version of this library are you using? The `release` version or the `latest` version?

amowry, 2023-07-14:
I'm using 5.0.2, which I believe is the release version.

lathoub, 2023-07-16:
Try to remove the release version (installed by the Arduino IDE) and replace with the latest version from github.

amowry, 2023-07-17:
That worked! Thanks!

lathoub, 2023-07-17:
@franky47 we need to make a release of the current code to avoid this in the future.

franky47, 2023-07-18:
@lathoub Agreed. I managed to fix the CI errors on the 5.1.0 PR, so it's in a good way for a release. I'll work on that further during my holidays next week.

shikshik, 2024-06-18:
Hi friends!
I'm receiving the same error when trying to change a single midi setting.
I'm using platformIO, and tried to pull directly the git maser branch as you suggested here and I still get the error.
I also tried the dev branches of v5.1.0 and "create_CUSTOM_INSTANCE" but again I get the same results.

Would appreciate your help, even if it's a quick fix I can do until this issue is resolved.

Thank you!

shikshik, 2024-06-18:
Update: I managed to use custom midi settings without the macros.