FortySevenEffects/arduino_midi_library #335: sendSysEx problam

komusics, 2023-10-09:
i have a problam when i type to send sendSysEx. i use arduino  1.6.10

my code.

#include <MIDI.h>

// Simple tutorial on how to receive and send MIDI messages.
// Here, when receiving any message on channel 4, the Arduino
// will blink a led and play back a note for 1 second.

MIDI_CREATE_DEFAULT_INSTANCE();

void setup()
{
    pinMode(LED_BUILTIN, OUTPUT);
    MIDI.begin(1);                      // Launch MIDI and listen to channel 1

    const byte * XGon[9] = {0xf0,0x43,0x10,0x4c,0x00,0x00,0x7e,0x00,0xf7};  
    MIDI.sendSysEx(9,XGon,false);
    
}

void loop()
{
    
}

after i verify my code Arduino ide has report problam.

[image]

how i cam fix ?

thankyou all.

komusics, 2023-10-09:
OK i found a problam with i add * in const byte  ( const byte * XGon[9]) .

becose no have a excemple to use this code.

closed 2023-10-09 by komusics

franky47, 2023-10-09:
In your first example, you created `const byte * XGon[9] = {...}`, which is an array of 9 pointers to a byte.

Since you likely want an array of 9 bytes, dropping the star gives you the correct type.

Tip: if the array never changes, consider moving it out of the `loop` function and mark it as `static const`, this will let the compiler optimise memory usage.