FortySevenEffects/arduino_midi_library #343: ESP32S3 Hairless MIDI on Serial

copych, 2024-03-24:
## Context

Please answer a few questions to help us understand your problem better and guide you to a solution:

- What board are you using ?
    - LOLIN S3 Pro
    - i2s DAC and a microSD card via SDMMC 
- What version of the Arduino IDE are you using ?
    - both 1.8.19 and 2.3.1
- How are you using MIDI ?
    - [ ]  Hardware Serial (DIN plugs)
    - [ ]  USB
    - [x]  Serial port of the board which corresponds to the UART0 and which is available thru the on-board USB type-c, I want Hairless MIDI to transmit data from PC to ESP32S3.
- Is your problem related to:
    - [x]  MIDI Input (reading messages from other devices)
    - [ ]  MIDI Output (sending messages to other devices)
- How comfortable are you with code ?
    - [ ]  Complete beginner
    - [x]  I've done basic projects
    - [x]  I know my way around C/C++
    - [ ]  Advanced / professional

## Describe your project and what you expect to happen:
I have tried all the ways of starting a MIDI IN on a Serial 115200 for Hairless MIDI. But no luck. Now I use the current version of the library downloaded from here.

## Describe your problem (what does not work):

## Steps to reproduce
for example, this code won't compile, nevertheless I have tried variants that compile but MIDI.read() kept quiet.
```
struct CustomBaudRateSettings : public MIDI_NAMESPACE::DefaultSerialSettings {
  static const long BaudRate = 115200;
};

#if defined(ARDUINO_SAM_DUE) || defined(USBCON) || defined(__MK20DX128__) || defined(__MK20DX256__) || defined(__MKL26Z64__)
    // Leonardo, Due and other USB boards use Serial1 by default.
    MIDI_NAMESPACE::SerialMIDI<HardwareSerial, CustomBaudRateSettings> serialMIDI(Serial1);
    MIDI_NAMESPACE::MidiInterface<MIDI_NAMESPACE::SerialMIDI<HardwareSerial, CustomBaudRateSettings>> MIDI((MIDI_NAMESPACE::SerialMIDI<HardwareSerial, CustomBaudRateSettings>&)serialMIDI);
#else
    MIDI_NAMESPACE::SerialMIDI<HardwareSerial, CustomBaudRateSettings> serialMIDI(Serial);
    MIDI_NAMESPACE::MidiInterface<MIDI_NAMESPACE::SerialMIDI<HardwareSerial, CustomBaudRateSettings>> MIDI((MIDI_NAMESPACE::SerialMIDI<HardwareSerial, CustomBaudRateSettings>&)serialMIDI);
#endif
```

I try to receive midi data from the PC to the ESP32S3 via Hairless MIDI as I used to some time ago, but I can't. I only manage to run classical MIDI IN on a selected RX pin, so I also have to run  some kind of a media convertor on a Arduino Micro Pro, which is actually is an unneeded extra in my case.

lathoub, 2024-03-24:
You are probably using the Release version (Arduino Lib manager). Using custom baud rates require the latest code (either fork or download the latest)

copych, 2024-03-24:
I use the version downloaded from this repo, from the master branch.
Is there a 100% working code for an ESP32S3 Hairless Midi on Serial that someone has successfully run on a certain versions of IDE and ESP Arduino core? I would reproduce the environment and see if it works for me.
BTW, what is the correct Serial name for ESP32s3 board? When I use 'Serial' it ruins during compilation. The best that I could imagine was 'Serial0', this way it compiles but no events received (((

copych, 2024-03-26:
@lathoub 
Update: I have finally managed to run Hairless MIDI. The problem is somewhere in the board definition file(s). I haven't investigated it deeply, but as I have changed the board from the "LOLIN S3 Pro" to the generic "ESP32S3 Dev Module", everything worked fine. Sorry for bothering.

closed 2024-03-26 by copych