FortySevenEffects/arduino_midi_library #345: MIDI.sendPitchBend() Problem

matalog, 2024-03-27:
The description of MIDI.sendPitchBend() here [https://fortyseveneffects.github.io/arduino_midi_library/a00032.html#gaaaf16c1334640e690ecbd835d645e4cb](url) is not correct.  The negative values of the float f are not converted to negative numbers - they end up positive, so a sweep of the full range of pitch bend values with float from -1 to 1 results in an initial higher bend down to zero and back up to a higher bend again, instead of an initial low bend to zero to high bend.

I was able to use the MIDI.sendPitchBend() correctly by leaving out your code, and scaling my values to signed 16 (14) bit integers and putting that directly in MIDI.sendPitchBend(s14int,1)  for example.

marchingband, 2024-05-18:
Same here, I am using ESP32-S3, not sure if this is an arch related bug. I noticed `MIDI_PITCHBEND_MAX` is `8191`, perhaps if it was `8191.0` it would work as intended.

crossan, 2025-04-19:
The code has a bug, it attempts to scale the -1 to 1 range from -8192 to  8191, however the scale value it calculates is negative for the negative range, when it should always be positive.
Multiplying negative input by this negative scale value makes its values positive.