FortySevenEffects/arduino_midi_library #346: Issue with Ableton and midi "feedback"

nicolasvair, 2024-04-17:
Hi,

This was first posted in the BLE-midi repo [here](https://github.com/lathoub/Arduino-BLE-MIDI/issues/87).

First thank you for this fantastic repo. I'm doing a "4 knobs" wireless controller using encoders and led ring.

[image]

The idea is to be able to control a parameter from various devices and have led feedback on each of them. The other being [midi fighter twister](https://store.djtechtools.com/products/midi-fighter-twister) is also based on encoders and led rings.

I came across a strange behavior, if arduino is connected to computer through BLE-midi, I have a strange behavior on ableton parameters mapped to midi CC. Not sure how to describe it, it is jerky.

https://github.com/FortySevenEffects/arduino_midi_library/assets/79313360/c457c615-6e45-4810-8048-084f8377c904

Important to note, this is independent of my code. I can upload a bare minimum code for connecting, and still have the issue. Even without :
`MIDI.setHandleControlChange(OnControlChange);
`
For example I have the issue with this code :

> #include <BLEMIDI_Transport.h>
> //#include <hardware/BLEMIDI_ESP32_NimBLE.h>
> //#include <hardware/BLEMIDI_ESP32.h>
> //#include <hardware/BLEMIDI_nRF52.h>
> #include <hardware/BLEMIDI_ArduinoBLE.h>
> #include "Arduino_BMI270_BMM150.h"
> 
> BLEMIDI_CREATE_DEFAULT_INSTANCE()
> 
> unsigned long t0 = millis();
> bool isConnected = false;
> 
> void setup()
> {
> // MIDI INIT
> MIDI.begin();
> 
> pinMode(LED_BUILTIN, OUTPUT);
> digitalWrite(LED_BUILTIN, LOW);
> 
> BLEMIDI.setHandleConnected( {
> isConnected = true;
> digitalWrite(LED_BUILTIN, HIGH);
> MIDI.sendControlChange(1, 1, 1);
> });
> 
> BLEMIDI.setHandleDisconnected( {
> isConnected = false;
> digitalWrite(LED_BUILTIN, LOW);
> });
> 
> }
> void loop()
> {
> MIDI.read();
> }

Maybe related to how the arduino send a "confirmation" of the received CC from ableton ?
When I look into details, ableton sends values like : 24, 25, 27, 22, 29, 32, 35 ...

But it only happens with this device and this repo. I thought it may be related to bluetooth latency first, but I don't have any issue using TouchOsc on iPhone... so lathoub advised to post here.

Any hints where to look at ?

Thank you 🙏

franky47, 2024-04-25:
Could you try a wired connection (MIDI DIN or USB) to rule out an issue in the Bluetooth stack?

nicolasvair, 2024-04-25:
I tried USB by adding : 
#include <USB-MIDI.h>
USBMIDI_CREATE_DEFAULT_INSTANCE();

But I get the following errors : 
-  #error MIDIUSB can only be used with an USB MCU.
and #error "Unsupported architecture"

I'm using a arduino nano ble sense, it seems that it can't act as general compliant usb device (not sure I understood all of it).

From what I checked rapidly, using DIN midi would require buying octocouplers and electronics.. 

Do you have any other idea how to try usb or din ?

Or maybe ther ie another test that I could do before diving into this ? A "debug" or "verbose" to check what packages are sent and come back but at a lower level than midi ? 

I suspect it is the "confirmation" of received message sent by computer or midi device but I don't know if it works like this.

nicolasvair, 2024-05-08:
Hi, 

So it seems to be related to midi thru...

I'm not sure if the issue is related to BLE-midi or forty-seven

If I try a   MIDI.turnThruOff(); before or after   MIDI.begin(MIDI_CHANNEL_OMNI); it doesn't disable the thru.

My post in BLE-midi is here
https://github.com/lathoub/Arduino-BLE-MIDI/issues/87

It is a real bummer because Ableton sees it as a feedback loop and disable all automation...

franky47, 2024-05-08:
Indeed, Thru doesn't make sense for non-serial transports. @lathoub I remember you added a way to dictate whether Thru was enabled or not per transport?

lathoub, 2024-05-09:
>I remember you added a way to dictate whether Thru was enabled or not per transport?

Correct,  `thruActivated` is set per library. Serial MIDI (this library)

https://github.com/FortySevenEffects/arduino_midi_library/blob/2d64cc3c2ff85bbee654a7054e36c59694d8d8e4/src/serialMIDI.h#L54

In packet based overrides (aka packet based protocol), `thruActivated` is set to false;

https://github.com/lathoub/Arduino-AppleMIDI-Library/blob/e0abb86dee1c7cc09574eae3e3fad09fd4785810/src/AppleMIDI.h#L151

https://github.com/lathoub/Arduino-BLE-MIDI/blob/929c2fc04962672ddba903a618601bea44ee1f3e/src/BLEMIDI_Transport.h#L55

So in BLE MIDI, MIDI Thru is disabled, and MIDI notes should not be echoed.