FortySevenEffects/arduino_midi_library #350: Midi note input gets reflect on midi out

amritgenius, 2024-07-06:
im trying to get midi note on/off on an Arduino pro micro and turn on/off leds according to note on off and pitch, its working but whatever the Arduino gets over midi in port it send also to midi out port, this is calling feedback loop or something, i have tried also arduino uno instead of pro micro, same story.

franky47, 2024-07-06:
This is called [soft Thru](https://fortyseveneffects.github.io/arduino_midi_library/a00035.html), and you can turn it off this way (in your setup function):

```c++
MIDI.turnThruOff();
```

amritgenius, 2024-07-06:
void setup() {
 MIDI.turnThruOff();
  MIDI.begin();
}

i was using in this order and that was not working. Now with your code its working well. Thankyou so much.

closed 2024-07-07 by amritgenius