FortySevenEffects/arduino_midi_library #353: MIDI.turnThruOff(); not working

AaronHaCohen, 2024-08-10:
Thanks for providing this library

- What board are you using ?
Arduino Uno R3
Arduino IDE 2.3.2
- How are you using MIDI ?
hardwired to Rx & Tx from controller to MIDI monitor
- Is your problem related to:
    - [ ]  MIDI Input (reading messages from other devices)
    - [x ]  MIDI Output (sending messages to other devices)

MIDI.turnThruOff(); does not turn off MIDI thru

- How comfortable are you with code ?
60 years with embedded ASM, plain C  and some Fortran
Not very experienced with C++
Retired EE, designed and built data acquisition systems.

## Describe your project and what you expect to happen:

Using your Callback.ino example I added MIDI.turnThruOff(); at line 40 in setup but it seems to be ignored. Both original example and with MIDI.turnThruOff() added,  compile, upload and run without error. In both cases  all incoming MIDI in is sent out. Midi thru is not turned off.
MIDI.turnThruOff();  was copied directly from your Chaining.ino (but unchained)
Curiously your Chaining.ino example will not compile. It throws an error message: "invalid use of void" and the cursor stops at line 10.

Is the syntax of MIDI.turnThruOff();  wrong?

Note: Attachments (circuit diagrams, code examples) are most welcome and will help us understand your needs better and find a suitable solution for your issue.

--> Callbacks2.ino included below

## Describe your problem (what does not work):
MIDI.turnThruOff(); does not turn MIDI thru off

## Steps to reproduce
After uploading Callbacks2.ino
Connect a MIDI controller to Rx of UNO and a MIDI monitor to Tx.
Send any MIDI to the UNO and all incoming messages are sent out to the monitor, tho no messages are sent by the explicit code functions.

lathoub, 2024-08-10:
Discussion #351 #352

lathoub, 2024-08-10:
Make sure `turnThruOff()` is called after `begin()`

https://github.com/FortySevenEffects/arduino_midi_library/blob/2d64cc3c2ff85bbee654a7054e36c59694d8d8e4/src/MIDI.hpp#L1437-L1444

(`begin()` set Thru to true for serial MIDI)

https://github.com/FortySevenEffects/arduino_midi_library/blob/2d64cc3c2ff85bbee654a7054e36c59694d8d8e4/src/MIDI.hpp#L96-L97

The `thruFilter` is called at the end of the `read() `function

https://github.com/FortySevenEffects/arduino_midi_library/blob/2d64cc3c2ff85bbee654a7054e36c59694d8d8e4/src/MIDI.hpp#L822

And the thruFilter function starts with checking if thru is activated

https://github.com/FortySevenEffects/arduino_midi_library/blob/2d64cc3c2ff85bbee654a7054e36c59694d8d8e4/src/MIDI.hpp#L1455-L1460

Add a debug print of turn, LED on, ... at line 1461 to check if code execution gets this far

franky47, 2024-08-10:
What version of the MIDI library do you have? You can usually see this in the Library Manager in the Arduino IDE.

AaronHaCohen, 2024-08-10:
> What version of the MIDI library do you have? You can usually see this in the Library Manager in the Arduino IDE.

I am using 5.0.2  Thanks

AaronHaCohen, 2024-08-10:
It was suggested that I include the code I'm using to help solve this problem. I did include it between the <> characters in my original post but it seems to have disappeared.  I am posting it again here in case it is helpful. It is simply the Callbacks.ino example with MIDI.turnThruOff(); copied directly from your Chaining.ino example (but unchained) and inserted at line 40.  I hope this helps. Thanks, Aaron

` // Callbacks2.ino
// added MIDI.turnThruOff(); at line 40
#include <MIDI.h>

MIDI_CREATE_DEFAULT_INSTANCE();

// -----------------------------------------------------------------------------

// This function will be automatically called when a NoteOn is received.
// It must be a void-returning function with the correct parameters,
// see documentation here:
// https://github.com/FortySevenEffects/arduino_midi_library/wiki/Using-Callbacks

void handleNoteOn(byte channel, byte pitch, byte velocity) {
  // Do whatever you want when a note is pressed.

  // Try to keep your callbacks short (no delays ect)
  // otherwise it would slow down the loop() and have a bad impact
  // on real-time performance.
}

void handleNoteOff(byte channel, byte pitch, byte velocity) {
  // Do something when the note is released.
  // Note that NoteOn messages with 0 velocity are interpreted as NoteOffs.
}

// -----------------------------------------------------------------------------

void setup() {
  // Connect the handleNoteOn function to the library,
  // so it is called upon reception of a NoteOn.
  MIDI.setHandleNoteOn(handleNoteOn);  // Put only the name of the function

  // Do the same for NoteOffs
  MIDI.setHandleNoteOff(handleNoteOff);

  // Initiate MIDI communications, listen to all channels
  MIDI.begin(MIDI_CHANNEL_OMNI);
  // ***********    line 40 added  ***********
  MIDI.turnThruOff();
}

void loop() {
  // Call MIDI.read the fastest you can for real-time performance.
  MIDI.read();

  // There is no need to check if there are messages incoming
  // if they are bound to a Callback function.
  // The attached method will be called automatically
  // when the corresponding message has been received.
}
 `

AaronHaCohen, 2024-08-10:
Lathoub wrote:
"(begin() set Thru to true for serial MIDI)"
and
"Add a debug print of turn, LED on, ... at line 1461 to check if code execution gets this far"

Thanks for all your help. Are you suggesting that I edit my copy of MIDI.hpp to test its behavior?

I'm very inexperienced with C++, having only work with ASM and plain C for most of my career. It may take me a good while to learn enough to do this as well as to understand it all.
How do I "Add a debug print of turn, LED on,"?
And print to Arduino IDE serial monitor?

Thanks again for all your efforts, Aaron

AaronHaCohen, 2024-08-10:
Sorry for closing and reopening. I very new to this github system

lathoub, 2024-08-11:
> Are you suggesting that I edit my copy of MIDI.hpp to test its behavior?

yes

(Surround the code with triple ` for [code blocks](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code))

> How do I "Add a debug print of turn, LED on,"?
> And print to Arduino IDE serial monitor?

Yes

AaronHaCohen, 2024-08-11:
turnThruOff is now working.  When I went to edit the MIDI.hpp file as you suggested I discovered line 1461 was not the same as your code. Further investigation showed that the file was not the same. Seems that when the IDE updated the MIDI library to 5.0.2 a while back it didn't quite do it right.  So I deleted the library and reinstalled it and now MIDI.turnThruOff works. Thanks for all your efforts. Aaron

franky47, 2024-08-12:
Wow, that's some great sleuthing here! I would not have expected the IDE itself to cause an invalid source file.

Congrats for fishing this out, and thanks @lathoub for your guidance.

closed 2024-08-12 by franky47

AaronHaCohen, 2024-08-12:
Hello,
 I appreciate the compliment but don't really deserve it. It's really
thanks to Bart's suggestion to edit the MIDI.hpp to set a watchdog at line
1461. When I opened my library file it certainly didn't look like Bart's
example at line 1461. I never would have suspected it since the IDE told me
I was using version 5.0.2.
I'm now getting marginally competent at these C++ ways of doing things.
Thanks again, Aaron