FortySevenEffects/arduino_midi_library #359: V5.1.0 Coming?

woodenplastic, 2024-12-26:
Hi,
As im.very interested in Thru Maps i just wanted to ask how its going?
I tried the dev branch but i get a lot of Compiler errors so i guess this is still completly in progress?

Best
Adrian

franky47, 2024-12-26:
What kind of errors did you get?

Unfortunately, between work and web OSS in [@47ng](https://github.com/47ng), I don't have a ton of free time to give to this project anymore.

woodenplastic, 2024-12-26:
Ok, falls understand. Seeing how many people build from this lib i would love to put some effort in it if possible. 
I'll put together my problems in a list so we would be able to address them, ive got some hope lathoub would be in for help.
Thx for your quick response anyways!

woodenplastic, 2025-01-01:
Hi,

for example when i just try the example for Filters, having the lib defined in platform io like that:
https://github.com/FortySevenEffects/arduino_midi_library.git#feat/v5.1.0

i cant even start because MIDIMessage is not defined. what do i miss here?

franky47, 2025-01-01:
Hum that's weird, the definition should be here:

https://github.com/FortySevenEffects/arduino_midi_library/blob/7e9222b4634cdc510f7b76e4c356dfb25f69b174/src/serialMIDI.h#L109

Does it compile if you replace that `MIDIMessage` with `midi::MidiMessage<128>` ?

woodenplastic, 2025-01-01:
Hi, 
so i got it working. There is some syntax going on that i observed in my project before thats not intuitiv.
There is 3 Macros:
1. MIDI_CREATE_DEFAULT_INSTANCE();
2. MIDI_CREATE_INSTANCE(HardwareSerial, Serial1, DIN_MIDI0);
3. MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial1, DIN_MIDI0, Serial0MIDISettings);

Its working with 1. With 2 only if i do after that DIN_MIDI0Message. For me it was not possible to get it done with the custom instance.

My project also uses the other transports. 
APPLEMIDI_CREATE_INSTANCE(WiFiUDP, RTP_MIDI, config.hardware.devicename, 5004);
IPMIDI_CREATE_INSTANCE(WiFiUDP, IP_MIDI, 21928);

#ifdef BLE_ENABLED
BLEMIDI_CREATE_INSTANCE("STOMPCS", BLE_MIDI);
#endif

Here for example i have to do 
    BLEBLE_MIDI.setHandleConnected([]() {config.midiRouting.bleConnected = true;});
    BLEBLE_MIDI.setHandleDisconnected([]() {config.midiRouting.bleConnected = false;});
    
    but that is fine:
    
        BLE_MIDI.begin(MIDI_CHANNEL_OMNI);
    BLE_MIDI.turnThruOff();

    BLE_MIDI.setHandleNoteOn([](uint8_t channel, uint8_t note, uint8_t velocity) { m_onMidiReceive(4, 0, note, velocity, channel); });
    BLE_MIDI.setHandleNoteOff([](uint8_t channel, uint8_t note, uint8_t velocity) { m_onMidiReceive(4, 0, note, velocity, channel); });
    BLE_MIDI.setHandleControlChange([](uint8_t channel, uint8_t control, uint8_t value) { m_onMidiReceive(4, 2, control, value, channel); });
    BLE_MIDI.setHandleProgramChange([](uint8_t channel, uint8_t program) { m_onMidiReceive(4, 1, program, 0, channel); });
    
    It seems like the Macros are not what they could be. 
    I understand your point that you cant invest the time to fix a lot in this lib, but it would be lovely to give me a hint how to go about this so i can do some work in my fork that gives more consistency over all transports. 
    
    the other libs for example dont support 
    midiOutput.send(NoteOn, note, velocity, channel);
    
    for other things than Notes, CC and PC. 
    
    Thanks for your great lib without my project wouldnt be possible anyways!

franky47, 2025-01-06:
Ah it looks like the issue is in the BLE macro prepending `BLE` to the name you pass:
https://github.com/lathoub/Arduino-BLE-MIDI/blob/929c2fc04962672ddba903a618601bea44ee1f3e/src/hardware/BLEMIDI_ESP32.h#L172

cc @lathoub

woodenplastic, 2025-01-06:
Yes. I tried all macros and i had very different results.
Im using midi::message<128> now to be consistent for all interfaces but i struggle getting something out with:
Midi.send(msg).
Can it be that it doesnt work because this msg wouldnt be valid?
I commented out the valid check but still no output.
Also midi.send(type, data1, data2, channel) does not output something.

Is there anything i miss?