FortySevenEffects/arduino_midi_library #361: HardwareSerial2 pinout change till ESP32 3.0.0

bbaldach, 2025-01-16:
## Context

Please answer a few questions to help us understand your problem better and guide you to a solution:

- What board are you using ?
    - ESP32 DEV Module
    - dedicated Hardware board but standard (opto on Rx, 3,3v, etc..)
- What version of the Arduino IDE are you using ?
    - 2.3.4
    - ESP32 board 3.1.1 (but issue since 3.0.0)
- How are you using MIDI ?
    - [x ]  Hardware Serial (DIN plugs)
    - [ ]  USB
    - [ ]  Other (please specify)
- Is your problem related to:
    - [x]  MIDI Input (reading messages from other devices)
    - [x]  MIDI Output (sending messages to other devices)
- How comfortable are you with code ?
    - [ ]  Complete beginner
    - [x]  I've done basic projects
    - [ ]  I know my way around C/C++
    - [ ]  Advanced / professional

## Describe your project and what you expect to happen:

The board have to receive ProgramChange, Start and Stop messages to play MP3.

## Describe your problem (what does not work):

Since ESP32 3.0.0 , the MIDI doesn't work.
I have found that previous HardwareSerial pin had change. RX 16 -> 4  and TX 17 -> 25.
[Migration from 2.x to 3.0](https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html)
My board is already designed and I cannot modify it now.
is there a function to change new default pin to the previous one ? 

## Steps to reproduce
program compiled with ESP32 2.0.17 : working good
program compiled with ESP32 3.1.1 (globally version >= 3.0.0) doesn't work

franky47, 2025-01-16:
If I understand correctly, the software update remapped which pins are assigned to RX/TX for UART1 by default, so in order for your existing hardware to keep functioning, you'd have to explicitly map those to the right pins, using `setPins`.

Something that may be relevant in the v3 changelog:

> It is now possible to call setPins() before begin() or in any order.

bbaldach, 2025-01-16:
You're right. After several try , i just fixed by adding Serial2.setPins(16, 17, -1, -1); jst after the MIDI.begin();

```
#include <MIDI.h>

MIDI_CREATE_INSTANCE(HardwareSerial, Serial2, MIDI);

// -----------------------------------------------------------------------------
byte Bank = 0;
byte channel = 0;

void setup() {
  //pinMode(LED, OUTPUT);
  Serial.begin(115200);
  MIDI.begin();  // Launch MIDI, by default listening to channel 1.
  Serial2.setPins(16, 17, -1, -1);
}

void loop() {

  MIDI.sendNoteOn(42, 127, 1);    // Send a Note (pitch 42, velo 127 on channel 1)
  delay(500);		            // Wait for a second
  MIDI.sendNoteOff(42, 0, 1);     // Stop the note
  delay(500);		            // Wait for a second

  
}
```

closed 2025-01-16 by franky47