FortySevenEffects/arduino_midi_library #362: MIDI messages In come out on MIDI out

dblscore, 2025-02-15:
In wanting to create a tool that recovers the MIDI IN and transforms it (for example to make a transposition) I agree that systematically the translated MIDI OUT also sends the MIDI IN data. 
In the basic example (BASIC_IO), if I play a note 48 I get this scanning the TX : 

**17:39:37:533 Port1 Ch 1 NoteOn 90 30 0B [C2 ( 48), 11]**
17:39:37:534 Port1 Ch 1 NoteOn 90 2A 7F [F#1 ( 42), 127]
17:39:38:532 Port1 Ch 1 NoteOff 80 2A 00 [F#1 ( 42), 0]
**17:39:49:082 Port1 Ch 1 NoteOff 80 30 3A [C2 ( 48), 58]**
17:39:49:083 Port1 Ch 1 NoteOn 90 2A 7F [F#1 ( 42), 127]
17:39:50:081 Port1 Ch 1 NoteOff 80 2A 00 [F#1 ( 42), 0]

Note 48 (the one played as input) should not come out of TX. 
Is there a solution to isolate RX and TX?
Thank you very much
Ema

franky47, 2025-02-15:
You need to turn Thru off:

```
MIDI.begin();
MIDI.turnThruOff();
```

closed 2025-02-15 by franky47

dblscore, 2025-02-15:
Thanks a lot !!!!!!!!