FortySevenEffects/arduino_midi_library #367: Midi Library becomes unresponsive after "phantom" SysEx start byte.

submodify, 2025-09-06:
Hello,

Been having an issue that when hot swapping out TRS Midi cables it would occasionally cause the MIDI library to freeze or become unresponsive. 

It seems that hot swapping TRS cables can occasionally cause garbled or incorrect Midi data to be produced when plugging in or unplugging the cable (I'm sure DIN cables can do this to, on rare occasion, but it seems more common on TRS cables due to how they physically connect).

I've done some basic tests, but the problem seems to be that hot connecting a TRS connector, can on occasion create a phantom SysEx Start, or a SysEx Stop byte (seems to be more common when the TRS is already transmitting MIDI data when plugging in, especially if Clock messages are being sent). This will then seem to lock up the MIDI library (as its not really in the middle receiving a SysEx message).

When a SysEx Start byte is received, there doesn't seem to be any recovery when another Status Byte is received other than a SysEx End byte, which seems to cause the freeze.

I added this code between lines 1035 & 1036 of MIDI.hpp (which happens when the "extracted byte" is a Status byte), and it does seem to help:

```
        // If another Status byte is received after a SysEx Start byte
		if ((mPendingMessage[0] == SystemExclusiveStart)
		|| (mPendingMessage[0] == SystemExclusiveEnd))
			{
				// Something has gone wrong... Lets reset
				resetInput();
				return false;
			}
```
			

I'm a bit of a n00b at all this, so what I've done may not be suitable, or completely off, but it does seem to help (and I've not actually tested this with  SysEx messages). Anyway thanks for your work on this.

submodify, 2025-09-07:
Going by this information from:

[http://midi.teragonaudio.com/tech/midispec/sysex.htm#](url)

"The 0xF7 Status byte is dedicated to marking the end of a SysEx message. It should never occur without a preceding 0xF0 Status. In the event that a device experiences such a condition (ie, maybe the MIDI cable was connected during the transmission of a SysEx message), the device should ignore the 0xF7.

Furthermore, although the 0xF7 is supposed to mark the end of a SysEx message, in fact, any status (except for Realtime Category messages) will cause a SysEx message to be considered "done" (ie, actually "aborted" is a better description since such a scenario indicates an abnormal MIDI condition). For example, if a 0x90 happened to be sent sometime after a 0xF0 (but before the 0xF7), then the SysEx message would be considered aborted at that point. It should be noted that, like all System Common messages, SysEx cancels any current running status. In other words, the next Voice Category message (after the SysEx message) must begin with a Status."

I think there should also be a check somewhere in MIDI.hpp to ignore an 0xF7 (SysEx end) status byte, if the pending message hasn't received an 0xF0 (SysEx Start) byte. After a quick look it doesn't currently seem the case.

... and athough my code snippet above does mitigate the issue by aborting the SysEx message if a Channel Status byte is received, I guess it should also start a new pending message using that new Channel Status byte, and not just discard it.

I may have a go myself at sorting this out, but my coding skills are average at best :)

lathoub, 2025-09-07:
> I may have a go myself at sorting this out, but my coding skills are average at best :)

Good opportunity to better your coding skills! 
If you have working and tested code, pls issue a PR.

mink99, 2025-09-10:
Just to add some „historic“ knowledge and experience.
When the trs wiring for midi was introduced, there was a good reason that it would not be part of the standard. When you plug in a trs cable, the one connector of the plug (the tip) would make connections to all 3 contacts of the sink. This would create several random bytes to be recognised by the serial receiver. 
To avoid this the din connector has clearly separated the signal, pwr and gnd pins, so that only the correct connectors touch each other.
When Arturia released the beatstep they argued that using mini trs would reduce the height of the device in comparison to a proper din midi plug, which was proven an utter lie , by simply measuring. But it created just these problems as described, random garbage when hot plugging.
The same applies for example on behringer mini synths if you „hot plug“ it by pulling and pushing the mini-trs.
But there is a simple solution : just use an adapter from trs to midi-din wiring, and always keep the adapter attached to the device, and „hot plug“ the midi cable.
Even mechanical switching solutions like the Monacor midi patchbay (mme80) work like that without issues.
I doubt that it is worth the effort, to harden any library against random garbage, when a simple adapter or connection box will solve that issue much cheaper and more reliable.

submodify, 2025-09-10:
Garbage data is fine, as long as it does cause the "Device" to stop functioning as it currently is with the garbage SysEx Start/Stop byte. 

I've tested other commercial Midi devices (Like CME) an the 'Garbage' produced is very similar, although they are obviously handling the SysEx side properly as they don't seem to lock up.

submodify, 2025-09-10:
> You could (albeit with a very low probability) catch the end of a partial SysEx stream mid-flight when hot-plugging a DIN cable, so this at least is worth handling (IIRC it already does, by ignoring orphan data bytes not associated with an initial status byte).

I'm not sure what would happen with the data here. Every time I look at the current parse() function it confuses me. It looks to me that a SysEx Stop byte is treated exactly the same as a SysEx Start, so a lone SysEx Stop byte in the above situation (after plugging in a DiN mid SysEx) would start a new SysEx message which is totally bonkers.

My rewritten parse() function above should handle this properly and just discard the SysEx Stop..... and is a little less bloated than the original  :)

submodify, 2025-09-13:
So I've made a bit of a change to parse() that I wouldn't mind getting some opinions on (I've updated the code in my previous post).

So currently, if a SysEx message is larger than than the buffer size, it is split into buffer sized chunks like so...

The first chunk will start with 0xF0 (SysEx Start) and end with 0xF0 (SysEx Start)....
Any chunks in the middle will start with 0xF7 (SysEx End) and end with 0xF0 (SysEx Start)....
and the last chunk will start with 0xF7 (SysEx End) and end with 0xF7 (SysEx End)....

Every time the buffer is full, parse() will insert the appropriate status bytes to comply with the above, notify the SysEX callback to deal with it, prep the next chunk, and then parse() will return false and wait for more data.

On completion of the final chunk, the SysEx message is marked as a valid message, and parse() will return true, as it would do for a non split SysEx message. The problem is now, is that the SysEx Array for this last chunk now starts with a 0xF7 (SysEx End), and the message is not a complete message, so can it really be considered as a valid SysEx Midi message now?

Heres the definition from the library
```cpp
/*! This boolean indicates if the message is valid or not.
     There is no channel consideration here,
     validity means the message respects the MIDI norm.
     */
    bool valid;
```

I've made some changes to my above post, so that on completion of the final SysEx chunk, instead of parse() marking it as valid, and returning true, it will mark the message as not valid, then notify the callback to deal with it, return false, and simply move on to the next message.

To me, this seems like the best way to deal with SysEx overruns, otherwise for large SysEx messages (that you may not be expecting) parse() will return true, but at that time will only give you a partial Sysex message that starts with 0xF7 (SysEx End).

Basically I think if a SysEx message overruns its buffer and has to be split, it can no longer be considered a valid Midi message any more, and should then be left exclusively to callbacks to deal with.

I'd be grateful for opinions on this.

submodify, 2025-09-14:
Tested Running Status, and SysEx (including split SysEx messages) this morning and all seems good. Had one minor little issue, but was an easy fix. I used the following function (via the SysEx callback) to test the SysEx messages (both complete & split). You can also now use check() to see if the SysEx is valid (within the buffer size) or split (overrun).
```cpp
// Retransmit SysEx
void handleSysex(byte* arr, unsigned len) {
  // Check first if this is a valid sysex message
  if (MIDI_A.check()) {  // Complete SysEx
    MIDI_A.sendSysEx(len, arr, true);
  } else {                                   // Split SysEx
    if (arr[0] == 0xF0) {                    // First Chunk F0.....F0
      MIDI_A.sendSysEx(len - 1, arr, true);  // Send all but last byte (F7)
    } else {
      if (arr[len - 1] == 0xF0)  // Second Chunk F7.....F0
      {
        MIDI_A.sendSysEx(len - 2, arr + 1, true);  // Cut first and last bytes
      } else {                                     // Must be last Chunk F7.....F7
        MIDI_A.sendSysEx(len - 1, arr + 1, true);  // Send all but first byte (F7)
      }
    }
  }
}
```
I set my Max SysEx buffer to 16, and it didn't seem to have any issues retransmitting large (632 byte) messages. I was using a Teensy 4.0 for this test.

I guess I'll create a PR soon, although one thing I'm unsure about is how to do the unit test. My coding skills are average at best (mostly self taught in my spare time), and I'm only just about getting my head around Git, so how to setup the unit test (or create a new one for that matter) might be a bit beyond me at the moment =)

robmurru, 2026-10-02:
Same bug here, on two different ATmega328P MIDI modules (Befaco MIDI Thing and a DIY 1-to-8 thru), both on 5.0.2 and driven from a Beatstep Pro. Clock keeps flowing while notes, CC and All Notes Off are dropped on every channel until power cycle.

The mechanism, for anyone landing here: once `mPendingMessage[0]` is F0, every status byte other than Real Time or F0/F7 falls into `default: break;` in the status switch of `parse()` and is stored into `sysexArray` as data. When the buffer fills, the split branch emits the chunk and re-enters SysEx state, so the parser never recovers by itself. Real Time is delivered before the state machine, which is why clock survives. A single orphan F0 is enough. The Beatstep Pro merges its USB input onto DIN OUT, so a truncated editor dump will do it, and so will one byte corrupted at the optocoupler.

**Quick test on a stuck device:** send any complete SysEx. Its F7 frees the parser and notes come back immediately.

The minimal fix I'm using goes in that `default:` case:

```cpp
default:
    if ((mPendingMessage[0] == SystemExclusiveStart)
    ||  (mPendingMessage[0] == SystemExclusiveEnd))
    {
        mLastError |= 1UL << ErrorParse;
        if (mErrorCallback)
            mErrorCallback(mLastError);

        resetInput();
        return false;
    }
    break;
```

It drops the status byte that aborted the SysEx, so one message is lost. #368 goes further and starts a new message from that byte, which is closer to the spec. Either would be a big improvement over the current behaviour. I've been running this patch on the MIDI Thing for about two weeks with no recurrence. Patch and context: Befaco/midithing#19.